horizon-eda/horizon #124: Package Editor: Auto increment duplicated Padnames

atoav, 2018-03-04:
If I duplicate a Pad in the Package Editor right now, it gets the exact same name as the source of the duplicate right now. I propose a "smart" autonaming function, that counts up in integers.  

Say I have a pad called `6` and I duplicate it (and there is no pad called `7`), then name the duplicate `7`, because that should be the most typical usecase. If I have a Pad called `Input` and I duplicate it, call it `Input 2` etc.  

This is just a minor problem, so nothing urgent

carrotIndustries, 2018-03-30:
Incrementing single pads should be trivial to implement. What's your expectation when copy/pasting multiple pads?

atoav, 2018-03-30:
I wonder. I could imagine that this could get out of hand very easily. Therefore I propose a solution with two steps:
1. Copying/Duplicating pads increment the integers if the names are integers, if the names are something else, I propose some kind of numeric suffix like `foo 2` or something like that
2. A tool to rename selected pads based on some sorting system

Let me elaborate this idea a little more:  
When working on the DIP packages I would typically draw one package with say 8 pins, with all layers needed. Then I would duplicate the package and call it 10 pins, extend it to the size of the 10-Pin package, move padds, duplicate the missing two pads and rename them. Then I would recenter the package, and save it.  

What would be very cool for similar task would be a tool where I can basically rename a bunch of pads at once based on directions (e.g. left to right, top to bottom, clockwise, counterclockwise). That way you would have only to get positions right, and then run `number pads` and and check if you like what you got. I don't really know how you deal with different positions of pad 1 etc, but maybe you have a clever idea?