horizon-eda/horizon #126: Logo not displayed in ubuntu taskbar

greenscreenflicker, 2018-03-12:
[image]
We have such a nice logo now. However in ubuntu it is not displayed.

carrotIndustries, 2018-03-14:
That may be due to toe ubuntu panel trying to load the icon from the desktop file which isn't existing right now. As soon has we have properly packaged releases, this should go away.

atoav, 2018-03-21:
[image]  

I have a similar problem in Windows 10 (Started `horizon-pool-mgr.exe` and pinned it to the Task Bar). As you can see the Windows AeroPeek thingie displays the icon while the taskbar forgets about it as soon as you pin it).

atoav, 2018-04-08:
For Windows 10 this issue has been fixed with d51f34a6b6ded83427ff2bbeef7a92c3b799b11c
Cannot tell wether this is true for Ubuntu

greenscreenflicker, 2018-05-02:
@carrotIndustries @atoav Confirm fixed in ubuntu.

closed 2018-05-02 by greenscreenflicker