horizon-eda/horizon #127: Build fail on 16.04

popey, 2018-03-19:
Hullo! I'm trying to build on 16.04 with the gcc7 from the ubuntu toolchain ppa and gnome 3.26 from the ubuntu desktop team ppa. I think I have all the right dependencies, having followed the building on linux guide on the wiki, but I get a build failure..

```g++ -c -Isrc -I3rd_party -g3 -D_USE_MATH_DEFINES -fdata-sections -ffunction-sections -pthread -I/usr/include/uuid -I/usr/include/gtkmm-3.0 -I/usr/lib/x86_64-linux-gnu/gtkmm-3.0/include -I/usr/include/atkmm-1.6 -I/usr/include/gtk-3.0/unix-print -I/usr/include/gdkmm-3.0 -I/usr/lib/x86_64-linux-gnu/gdkmm-3.0/include -I/usr/include/giomm-2.4 -I/usr/lib/x86_64-linux-gnu/giomm-2.4/include -I/usr/include/pangomm-1.4 -I/usr/lib/x86_64-linux-gnu/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib/x86_64-linux-gnu/glibmm-2.4/include -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/mirclient -I/usr/include/mircore -I/usr/include/mircookie -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/cairomm-1.0 -I/usr/lib/x86_64-linux-gnu/cairomm-1.0/include -I/usr/include/sigc++-2.0 -I/usr/lib/x86_64-linux-gnu/sigc++-2.0/include -I/usr/include/cairo -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng16 -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/imp/imp_schematic.cpp -o src/imp/imp_schematic.o
src/imp/imp.cpp: In instantiation of ‘horizon::ImpBase::run(int, char**)::<lambda(auto:2)> [with auto:2 = horizon::CanvasGL::SelectionQualifier]’:
/usr/include/c++/5/type_traits:2346:26:   required by substitution of ‘template<class _Fn, class ... _Args> static std::__success_type<decltype (declval<_Fn>()((declval<_Args>)()...))> std::__result_of_other_impl::_S_test(int) [with _Fn = horizon::ImpBase::run(int, char**)::<lambda(auto:2)>; _Args = {horizon::CanvasGL::SelectionQualifier}]’
/usr/include/c++/5/type_traits:2357:55:   required from ‘struct std::__result_of_impl<false, false, horizon::ImpBase::run(int, char**)::<lambda(auto:2)>, horizon::CanvasGL::SelectionQualifier>’
/usr/include/c++/5/type_traits:2361:12:   required from ‘class std::result_of<horizon::ImpBase::run(int, char**)::<lambda(auto:2)>(horizon::CanvasGL::SelectionQualifier)>’
src/imp/imp.cpp:229:95:   recursively required by substitution of ‘template<class T> void horizon::bind_widget(std::map<T, Gtk::RadioButton*>&, T&, std::function<void(T)>) [with T = horizon::CanvasGL::SelectionQualifier]’
src/imp/imp.cpp:229:95:   required from here
src/imp/imp.cpp:229:90: error: cannot call member function ‘void horizon::ImpBase::update_selection_label()’ without object
                                                   [this](auto v) { update_selection_label(); });
```

Here's the full log. https://paste.ubuntu.com/p/zPd3SWQ4jq/

Any ideas what I'm doing wrong?

rnestler, 2018-03-19:
Hi @popey building with Ubuntu 16.04 isn't supported (See the [wiki](https://github.com/carrotIndustries/horizon/wiki/Building-horizon-on-Linux), because the toolchain is too old there.

popey, 2018-03-21:
My bad, I was missing some libs. Got it working on Ubuntu 16.04. 
[image]

closed 2018-03-21 by popey

carrotIndustries, 2018-03-21:
Oh, that's good news. What did you have to install to get it to build? IIRC, the GCC included with ubuntu 16.04 isn't as smart as I'm expecting it to be.

popey, 2018-03-21:
Yeah, I built it in a clean lxc container to which I added the ubuntu toolchain ppa and gnome ppa from the ubuntu desktop team. I bundled it all together as a [snap](https://snapcraft.io/). If published in the snap store, that one snap will work on 14.04, 16.04, 17.10 & 18.04 (which releases in April). That basically covers all supported Ubuntu releases. It will also work on other distros too, if the user installs snap support.

It needs a bit more testing though. I was going to share it with a colleague who is into circuit design (I haven't done any for years) to exercise the snap a bit. Would be great to get this out to a wider audience.   Do you have any stable releases planned?

kasbah, 2019-02-05:
Hey, do you think the build works on 16.04 now, or is there a recent snap package I could try?

kasbah, 2019-02-05:
Ok, thanks. Will wait till I upgrade my machine. Great talk at FOSDEM by the way!