horizon-eda/horizon #128: [UI] Selfdocumenting Tooltips

atoav, 2018-03-21:
As for now I didn't find any tooltips implemented. I understand that this is probably a low priority task right now, because things are still moving quite a bit, but I like to put this here before I forget about it.  

I think _some_ solution for a more verbose description of things beeing hovered would make sense, as such a solution would make horizon and its functions more or less self documenting. This again happens with the goal of reducing friction and gotchas as much as possible.

The traditional solution would be to use tool tips, but it is by no means the only solution. The Audio software _Ableton Live_ for example has a dedicated help area in the lower left corner which displays help depending on the mouse hover:  
[image]
Useres can fold that help area away with one click and without menu-diving. Compared to tool tips this has some advantages:  
- Text length is not as limited as in traditional tooltips
- Help text is visible immidiately, no waiting needed
- Because a dedicated portion of the screen is used, the help texts won't cover other stuff by accident.

The disadvantages are:
- There must be space for this in the UI
- It is probably harder to implement than tooltips

carrotIndustries, 2018-03-23:
Tool tips are trivial to add in Gtk: Just call `->set_tooltip_text` on a widget. The main difficulty probably is coming up with helpful / non-tautological tooltip text.

peterus, 2018-04-06:
no idea how the ui's are designed or implemented, but with a good visitor pattern it should be very easy to implement (just a function call like the `->set_tooltip_text`.

atoav, 2019-12-21:
@carrotIndustries I volunteer to create tooltips for all the things, however I have no idea how and where I will be able to find the according piece of code that matches the thing I wanna annotate.

A smart thing would be to automatically display keyboard shortcuts for actions that have them as well, a bit like blender does it maybe:

[image]  

I think this could be certainly one of the more useful usability features.