horizon-eda/horizon #130: Clickable icon does not appear

NotAFile, 2018-03-29:
I think this is due to my window manager (i3), but horizon-prj-mgr does not show an icon in the top left like on the pictures.

carrotIndustries, 2018-03-29:
I'm using i3 as well and the application icon is there. Is the icon absent in other Gtk3 applications such as gnome-calculator?

NotAFile, 2018-03-29:
yes, gnome-calculator does not show it either

carrotIndustries, 2018-03-29:
Ok, so you may want to check `gtk-shell-shows-app-menu` and `gtk-decoration-layout` in your gtk `settings.ini`

NotAFile, 2018-03-29:
I don't have a settings.ini file

carrotIndustries, 2018-03-29:
Run `gtk-query-settings` and see for the aforementioned variables.
If `gtk-shell-shows-app-menu` is TRUE, add `gtk-shell-shows-app-menu=0` to the settings ini
If `gtk-decoration-layout` is doesn't contain menu, add `gtk-decoration-layout="menu:minimize,maximize,close"` to the settings ini

Then settings.ini goe in `~/.config/gtk-3.0/settings.ini`:

carrotIndustries, 2018-07-08:
closing due to inacticity

closed 2018-07-08 by carrotIndustries