horizon-eda/horizon #133: [UI] Arc Selectionarea can be too big

atoav, 2018-04-06:
The clickable area of arcs seem to be the bounding box of a full circle around the centerpoint. A click on the upper small arc will also propose to select the lower arc.  
[image]  

See this illustration for what happens vs what I would expect to happen. The cyan arc is our arc, the grey arc is the full circle and the dashed red line is the clickable area where the shape detects it's collision:  
[image]  

I think collision for arcs should be implemented as a segment of a donut like so:  This would avoid many cases where clicking one thing brings up an not so helpful selection of multiple similarily named things that you have to click in order to find out which it is:  
[image]

carrotIndustries, 2018-04-06:
Unfortunately non-square selection boxes aren't supported, so your last suggestion isn't feasible to implement for now.
decddab7846424cd9d9237d7e00eaa6abfab03a5 is an attempt at improving the situation with the least amount of changes by using the arc's bounding box

Fitting an angled rectangle around the arc should be the way to go for now, but I'll need to think some more on how to calculate angle and size from the arc.

atoav, 2018-04-07:
Oh that's unfortunate indeed.  

Fitting a rotated rectangle could maybe would work as such:
1. Calculate the Angle of the vector pointing from start to end point
2. Calculate the height `h` of the [circular segment](https://en.wikipedia.org/wiki/Circular_segment)
3. The magnitude of the vector is the rectangles width, the angle of the vector is the rectangles rotation and the height is defined by the formula.

Then you also need to adress how to flip it in the right direction (swapping the start and endpoint of the vector would do the trick). I wrote small 2D-vector library for python once, they can be really useful for that kind of stuff.