horizon-eda/horizon #139: Schematic Editor: $REFDES and $VALUE don't honor the position specified in symbol

atoav, 2018-04-16:
Let's say as an example, I specified the positions of `$REFDES` and `$VALUE` like so:  
[image]  

Once I place the part into the Schematic editor it will look like so (also the Symbol Editor's `Preview` displays it that way):  

[image]

carrotIndustries, 2018-04-20:
A symbol can have orientation-specific text placements. These are defined in the "Preview window". First of all, click clear for all orientations. For each orientation, place the texts as desired and click set in the preview window once the placement is okay.

The user interface in that regard is for sure awkward and unintuitive, but since only very few users are ever going to create symbols that require per-orientation text placements, I didn't spend a great deal of time at polishing the UI.

atoav, 2018-04-21:
AHA, very good to know! Means I will fix these symbols with per orientation text placement.  

From an UI standpoint I would suggest just adding that description you gave as text on top of the window if it doesn't cause too much work. E.g.:

_A symbol can have orientation-specific text placements. First click "Clear" for all orientations. Then for each orientation position the texts as desired and click "Set" in this window._ 

That is clearly much cheaper than a sophisticated UI solution, but helps to make clear what this Preview Window is for (which I didn't realize either). I thought it was for setting the zoom level for some not yet implemented preview in the Place Part Window..

commit fc7ac874 (2018-04-24): add explanatory text for symbol preview window

fixes #139
closed 2018-04-24 by carrotIndustries