horizon-eda/horizon #140: 3D-Viewer: Faces won't always display colors

atoav, 2018-04-16:
When importing a 3D model colors will not always work as expected. See this comparison between what I see in Freecad (or Alias Autostudio) and what the 3D-Editor shows:  

[image]   

I must say that I cant really get behind the issue, I checked the surface normals etc. Colors worked for other models, so I tried to reproduce where this error came from. I found, that the 3D-preview takes _some_ colors (the small slit below has the expected color):  

[image]  

A comparison of `check geometry` in Freecad shows that both are similar in type (I first suspected that one is a solid or a shell or something).  

1. The face that got the right color:  
```
Unnamed.Part__Feature100.Face1:
 VERTEX    : 4
 EDGE      : 4
 WIRE      : 1
 FACE      : 1
 SHELL     : 0
 SOLID     : 0
 COMPSOLID : 0
 COMPOUND  : 0
 SHAPE     : 10
```

2. Another face which is part of the same type just next to the one before:  
```
Unnamed.Part__Feature070_solid.Face1:
 VERTEX    : 20
 EDGE      : 20
 WIRE      : 4
 FACE      : 1
 SHELL     : 0
 SOLID     : 0
 COMPSOLID : 0
 COMPOUND  : 0
 SHAPE     : 45
```

Upon checking I found similar uncolored faces in other step files. The 3D-model pictured above can be found in the P0915N Pull Request here: https://github.com/carrotIndustries/horizon-pool/pull/32

carrotIndustries, 2018-04-20:
I'm afraid, but this issue is unlikely to get fixed in a timely manner since I don't completely understand the step importer (most has been pasted from kicad) and it's only a minor annoyance

atoav, 2018-04-22:
Step is a legacy mess. You can feel the dust on it. Internally you use some kind of polygonized format to display stuff with OpenGL if I am right? Maybe there is a more straightforward format option to additionally support (eg coming from Blender)?

carrotIndustries, 2018-04-24:
> Internally you use some kind of polygonized format to display stuff with OpenGL if I am right? 
Exactly, for rendering I only need a set of colored triangles defined like this: https://github.com/carrotIndustries/horizon/blob/master/src/util/step_importer.hpp#L29, so importing anything mesh-like should be trivial to add.

I previously took a look at assimp for doing so since that library supports a wide range of formats and is available in debian, arch linux and mingw, but considered importing STEP more important since the intended purpose of the 3D view is to get a quick idea of how thinks look before exporting the whole board as a STEP file.

closed 2019-01-20 by atoav