horizon-eda/horizon #141: Board Editor: Moving by enterining number won't move smashed $RD

atoav, 2018-04-17:
When I move a part with smashed `$RD` by entering new coordinates into the Position X or Position Y, the smashed `$RD` wont move with it, but stay at the place it was.

Expected behaviour: it should always stay relative to it's parents location and orientation.

Also: is there an easier way to re-align moved parts to the grid than setting it's coordinates to [0, 0] manually?

carrotIndustries, 2018-04-20:
> Expected behaviour: it should always stay relative to it's parents location and orientation.

For me it does so, cannot reproduce.

> Also: is there an easier way to re-align moved parts to the grid than setting it's coordinates to [0, 0] manually?

The cursor snaps to the center of parts, so move the cursor so that it snaps to the center of a part, then start the move tool and the cursor with the part attached will snap to the grid.

atoav, 2018-04-22:
See here, with this smashed part. When I move it via <kbd>m</kbd> it works just fine. 
As soon as I move it via entering a number and pressing <kbd>⏎</kbd> I only affect the part and not it's smashed `$RD`.
[image]  
I understand that could be good for some situations, but I think the behaviours of moving by <kbd>m</kbd> and moving via absolute value should be the same.

carrotIndustries, 2018-04-24:
Oh, I see. I thought you were referring to the move exactly tool.

commit a4087140 (2018-04-24): move texts when setting package position

fixes #141
closed 2018-04-24 by carrotIndustries