horizon-eda/horizon #154: Feature Request: Incremented Move with arrow keys

atoav, 2018-05-11:
In the recent days I was working a lot in the schematic editor and I found myself often in situations where I wanted to move a selection just by one grid unit. With the mouse this can be sometimes a bit of a hassle because your curser starts to lock onto things that are close by etc.  

It would be very cool if we could move things by one grid unit with the arrow keys. I can imagine this would also be quite useful for board and package editors:  
<kbd>←</kbd>: Move selected objects left by one grid unit
<kbd>→</kbd>: Move selected objects right by one grid unit
<kbd>↑</kbd>: Move selected objects up by one grid unit
<kbd>↓</kbd>: Move selected objects down by one grid unit
<kbd>Alt</kbd> + <kbd>←/→/↑/↓</kbd>: Move on finer grid  

Of course there might be other plans for the Arrow keys already, so just an idea.

carrotIndustries, 2018-05-11:
Seems easy to implement. Did you envision this to be part of the move tool or as a separate tool?

atoav, 2018-05-11:
Not sure. It should definitly be part of the move tool, but I could also imagine it would be even more useful if it would work without pressing <kbd>m</kbd> before and <kbd>Escape</kbd> after.  

So basically like a macro called by arrow key which just moves parts.

closed 2018-05-18 by carrotIndustries

atoav, 2018-05-18:
Cool I will give it a shot tomorrow

commit 8e6632cb (2018-05-18): move using arrow keys

fixes #154