horizon-eda/horizon #156: Schematic Editor: Annotates "Keep existing" option doesn't keep existing

atoav, 2018-05-15:
It seems to me that the Annotate Action ignores custom names for parts in some cases. For Audio Projects I like to name the jacks and potentiomers in a descriptive manner (e.g. `VOLUME`  instead of `RV3`). When I do that and run the annotate action `VOLUME` will be renamed automatically to `RVx` (with `x` as a counting integer).

carrotIndustries, 2018-05-16:
Having parts with non-numeric reference designator would mess up the generation of new reference designators. Apart from that, I've never seen anyone do that and don't think it's a feature worth supporting for now.

closed 2018-05-16 by carrotIndustries

atoav, 2018-05-17:
Might depend on context. I know that in an industry application a designer would never use non-standard reference designators – but I saw it quite often in amateur- or kit projects, where it can also make sense for readability. Don't get me wrong, I totally get where you're coming from with this. In a professional project nobody would do it that way – but good open source software pulls in all kind of people, and they expect the software to not get into their way without giving them a reason. Just yesterday I had a discussion about Blender. Coming from a Film/CG background I could give a million reasons why it makes sense things are structured the way they are, but that doesn't render the newbies complaints who are not _power users_ in the traditional sense invalid.

My main argument therefore would be that the expected behaviour of "keep existing" is, that everything that is manually set (exists) will be kept. Overwriting stuff that the user set (for which ever reason), is unexpected behaviour at this point. Maybe it can also be called "keep existing numeric reference designators" but that is a bit long and bulky...  

OT: Btw. I just saw your talk at GPN and I have to join the praise the guy at the end of the video recording gave you. What you managed to pull of there is really quite a thing already. I am no member of CCC but a long time lurker and I know a few people there (Hamburg), so if we should meet at one day I owe you a drink : )

carrotIndustries, 2018-05-18:
d30cff886a1bf9cc1c0605154a35ca2c8406597e adds the option to ignore non-numeric refdes during annotation

atoav, 2018-05-18:
Thanks a bunch 🥇