horizon-eda/horizon #158: Horizon as AppImage?

atoav, 2018-05-18:
A friend of mine had troubles with compiling horizon on a LTS version of ubuntu, he said maybe it would be a good idea to distribute it as a AppImage [as described here](https://www.booleanworld.com/creating-linux-apps-run-anywhere-appimage/) to make it easier to use on Linuxes with outdated packages.

carrotIndustries, 2018-05-18:
Some people at https://www.mikrocontroller.net/topic/417908#new managed to create an appimage. Maybe someone can integrate that into the existing CI

fk0815, 2018-05-18:
I prepared the build, it can be found here: https://build.opensuse.org/package/show/home:frank_kunz/horizon, but it fails for the AppImage for some time now: https://build.opensuse.org/build/home:frank_kunz/AppImage/x86_64/horizon/_log

funlw65, 2018-10-14:
Not an Appimage, but it fails a normal compilation under Ubuntu 18.04 LTS:

In file included from src/router/pns_horizon_iface.cpp:5:0:
src/canvas/canvas_gl.hpp:9:10: fatal error: glm/glm.hpp: No such file or directory
 #include <glm/glm.hpp>
          ^~~~~~~~~~~~~
compilation terminated.
Makefile:549: recipe for target 'src/router/pns_horizon_iface.o' failed
make: *** [src/router/pns_horizon_iface.o] Error 1

carrotIndustries, 2018-10-15:
Maybe you didn't install glm? The package is called `libglm-dev`.

funlw65, 2018-10-16:
Thank you for the library name, I hate the "-dev" separation!

atoav, 2018-10-16:
> Some random render artefacts when moving the mouse pointer over UI controls but not permanent, just temporary, appear like flashes... running it from KDE Plasma Window Manager...  
Is this by any change happening on the tooltips? Because I had that happen to me in the pool manager on Windows.

funlw65, 2018-10-16:
It does not happen now, it was at first run and yes, I think it was about
the Pool Manager (that is the first you work with, I guess...). I run into
part design, and also there was some slow motion in rendering (slow
refining of the graphic and chars, not sure), probably my system was
overloaded with applications at that time...

krasimirk, 2019-04-17:
Hi,
I have old Ubuntu 14.04 installed and the only way to try Horizon-EDA is via AppImage.
Is there a latest AppImage? 
The last one (the link above) was from 2018-02-15.

fk0815, 2019-04-19:
I tried to build a new AppImage, but the build fails for it:

[  338s] src/imp/imp_schematic.cpp: In constructor 'horizon::SelectSheetDialog::SelectSheetDialog(const horizon::Schematic*, const horizon::Sheet*)':
[  338s] src/imp/imp_schematic.cpp:687:9: error: 'class Gtk::ScrolledWindow' has no member named 'set_propagate_natural_height'; did you mean 'set_min_content_height'?
[  338s]      sc->set_propagate_natural_height(true);
[  338s]          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~

I'm using gtk 3.20.10 which has been working a year ago or so. Maybe in the meantime horizon requires a newer version. The full log can be found here: https://build.opensuse.org/build/home:frank_kunz/AppImage/x86_64/horizon/_log

The idea behind appimages is to build a software on an older distribution, pack all the dependencies into the AppImage and then it is able to run without installation on that release or newer/different ones. The AppImage I made is based on openSUSE leap 42.3, from mid of 2017. Ubuntu 14.04 is years older. To create a AppImage that can run on 14.04 it would require to build it on that release. I don't think that Ubuntu 14.04 will support the library versions that are needed for horizon.

carrotIndustries, 2019-04-19:
> I'm using gtk 3.20.10 which has been working a year ago or so. Maybe in the meantime horizon requires a newer version.

Yes, seems like we need 3.22 as of now. Is this a major showstopper for appimages that run on ubuntu 14.04? If yes, I could see if it's possible to do without features introduced in 3.22

krasimirk, 2019-04-19:
If you can make AppImage for Ubuntu 18.04, I can upgrade and try it.

krasimirk, 2019-04-20:
No need anymore AppImage. I installed Ubuntu 18.04, cloned the git repo and compiled/launch successfully.
Looks very nice!

carrotIndustries, 2020-10-08:
Since we have upstream-provided debian packages and flatpak, I see little need for an appimage.

closed 2020-10-08 by carrotIndustries