horizon-eda/horizon #159: Release TODOs

carrotIndustries, 2018-05-18:
After almost two years of off- and on development, I'm anticipating to do the first stable by the end of this year.

This issue is supposed do be the place for discussing which features should be part of this release.

Some suggestions in no particular order:

- [ ] Pool inheritance: #148 
- [ ] BOM export
- [ ] Improved schematic export (can search for text)
- [ ] Search/replace
- [ ] `$ATTR` text for symbols for displaying additional data such as voltage/power rating and tolerance on the schematic
 - [ ] Combined Symbol/Unit editor for simplifying the creation of new units and symbols
 - [ ] Pool manager should give 'where used' information
 - [ ] Preview for non-graphical pool items, i.e. list of pins instead of symbols
 - [ ] win32 installer (so far WiX seems the way to go)
 - [ ] Somehow unify layer display options and selection filter

pwnorbitals, 2020-10-24:
Probably outdated now ?

carrotIndustries, 2020-10-27:
I renamed the issue to better reflect what it's about. Some of the ideas like 
 -  Easier and faster (bulk) creation of hierarchical Parts that only differ in name, value etc.

are still valid.

unrelentingtech, 2022-01-19:
> A tool to (non-destructively?) round and/or bevel corners on traces

[Kicad 6 seems to do it destructively](https://www.youtube.com/watch?v=jUm6pz9fyeE). Non-destructive indeed would be great! Would be also amazing to have non-destructive teardrops.

> Something to help with stacked or multiple PCBs

When doing a split keyboard, I just did it side by side, and I was too lazy to split 3v3 and GND nets so there were two airwires between the halves and for exporting separately I just did the "delete half, export, undo, delete other half, export" dance. I wonder if I missed something again :) and if that's not the case, I'm looking forward to better multiple-boards-in-one-project support in the future someday..

> Colors for Schematic export

I looked into implementing this but was a bit confused about how colorscheme colors get into the rendering API. It doesn't seem easy currently to inject those colors into the PDF renderer (?)

---

Looks like a couple items can be checked off, such as PDF export line width and PDF board export.

carrotIndustries, 2022-01-19:
> how colorscheme colors get into the rendering API. It doesn't seem easy currently to inject those colors into the PDF renderer (?)

The first thing would be extending `Canvas::img_line` and `img_draw_text` and maybe some others to also get the `ColorP` color. We'd then need a place to map the palette colors to actual RGB colors. No idea if this should be a project or a user setting.

atoav, 2022-12-09:
> JLCPCB integration would be a huge plus (especially the parts library and export for assembly)

This is a matter of doing the following:

1. Create a mapping from MPNs to LCSC part numbers (this is non-trivial)
2. Remove parts that should not be assembled (through hole, parts unavailable at LCSC, ...)
3. For pick&place the packages need to be oriented the right way (sometimes the packages in the pool might be rotated 90, 180 or 270 degrees different than those LCSC uses)
4. Create the right format for their pick and place process (rename some headers in the CSV remove/reorder some columns, format certain numbers differently)

For my own projects I tend to write scripts like this one: [bom-to-jlcpcb.py](https://gist.github.com/atoav/7c2f931c64336b6b4259e115eef02f77)

That scripts removes certain parts from the BOM based on mpn, manufacturer and writes a new BOM. Then it does similar things with the pick and place file.

I can hardly see how one would make this part of horizon without constantly having to update the mappings between our parts and the sortiment of LCSC. Even for my own project I had to update the part numbers a month later because most of what I wanted was not available anymore. Now imagine doing that for the whole pool.

valpackett, 2022-12-12:
IIRC, whatever script KiCad users use has a whole bunch of hardcoded part orientations and stuff. It would be ideal if there was some shared community database of these, somehow listed based on something tool-neutral (particular features of the footprint? this *is* a difficult problem)