horizon-eda/horizon #161: Performance Issues: Part Browser gets very slow with big amount of parts

atoav, 2018-05-21:
Today I created multiple thousand resistor parts (see [horizon-pool/pull/48]https://github.com/carrotIndustries/horizon-pool/pull/48) and [horizon-pool/pull/49](https://github.com/carrotIndustries/horizon-pool/pull/49)) and this considerably slows down the part search and browsing. The pool is not getting smaller, so this is something that definitly needs optimization.  

I couldn't really figure out if the performance sink is the display code or the database part, but as soon as I manage to limit the query (e.g. by searching for more specific MPNs) it speeds up considerably.  

Working on a i7-6700K @4GHz with 16GB-RAM and a SSD

carrotIndustries, 2018-05-21:
914e854b3abe2015a6b7ac3a1df9e0d07a7f3528 tries to fix some of that by
 - using indexes and optimized queries
 - using search entries instead of regular entries (these wait some time for the user to type characters before emitting the event)

Search without tags should now be almost-acceptably fast and search with tags has gone from very slow to slow.

atoav, 2018-05-21:
Cool I will try it as soon as possible.

atoav, 2018-05-21:
Works very well! Thanks for the quick fix, amazing. Optimization is never done, but I think we can close this one.

closed 2018-05-21 by atoav