horizon-eda/horizon #162: Cannot open project because of missing part

atoav, 2018-05-21:
[image]  

Interestingly enough that file is in the cache, but I can't find `7ae4e182-4f7c-4287-8bba-90d8c45949bb` in top_sch.json, board.json or top_block.json, so I am not sure what the problem could be there.  
Deleting the cache file didn't help either. The project file can be [found here](https://code.hfbk.net/sdiy/invoff/tree/master/horizon/invoff).

carrotIndustries, 2018-05-21:
Do other/new projects encounter this issue as well? My guess is that there's another part in the cache that depends on 7ae4e182-4f7c-4287-8bba-90d8c45949bb. Try deleting that one.

atoav, 2018-05-21:
No such file in the cache files:  
[image]  

I realized it happens with _all_ my projects (some of which certainly don't contain `7ae4e182-4f7c-4287-8bba-90d8c45949bb`. When I create a new project it happens immidiately. I will restart my box and then try a fresh install (testing the latest build once it is finished), maybe that already fixes it.

atoav, 2018-05-21:
Okay I figured it out: I restarted and did a fresh install, and had still the same, even with a fresh clone of the project. The next logical step is to ask yourself whether the software has a storage that wasn't reset through the fresh install. So I renamed `%APPDATA%/Local/horizon` and it instantly worked again.  

I had a look at the json files and found that our infamous `7ae4e182-4f7c-4287-8bba-90d8c45949bb` was in the "part_favorites"-array in project-manager.json. Obviously the prj-mgr was looking for a favorite part that didn't exist anymore (because I made a new and tidy version of the resistors).  

**Fix:** We should consider ignoring missing part favorites without throwing errors on startup. Ideally without deleting them from project-manager.json to avoid deleting favorites when somebody stores their pool on a network drive that is temporarly unavialable.

carrotIndustries, 2018-05-21:
Thanks for discovering that, wouldn't have come to my mind that fast...

commit cdec500a (2018-05-21): ignore part favorite not found in pool

fix #162
closed 2018-05-21 by carrotIndustries