horizon-eda/horizon #163: Two binaries, one icon

atoav, 2018-05-21:
As for now we have two executables that have to be started seperately with the same icon. In the long run this needs to be solved in some way. As mentioned in #159 there are three obvious solutions:
1. the pool manager gets its own icon
2. the pool manager is to be opened from the prj-mgr (or the other way around)
3. a startup-window for both pool-mgr and prj-mgr  

Just because I had it lying around, here is a alternate icon draft. Any opinions?

| Name | Draft |
| ------ | ----- |
| Sky Pool 256x256  | [image]  |
| Sky Pool 64x64| [image]  |
| Sky Pool 32x32 | [image]  |
| Sky Pool 16x16 | [image] |

carrotIndustries, 2018-05-21:
>    the pool manager gets its own icon

From all options presented, this seems to be the most sensible one.

>   the pool manager is to be opened from the prj-mgr (or the other way around)

Doesn't make much sense TBH.

>    a startup-window for both pool-mgr and prj-mgr

As both (espescially the project manger) are some kind of launcher themselves, that would be kinda awkward.

atoav, 2018-05-22:
> From all options presented, this seems to be the most sensible one.
Okay then we're on the same page, I also liked this variant the most. Like last time I will try to produce a few drafts for the pool managers logo.  

I will try to:
- keep it visually and color wise very close to the other icon (so it is clear they belong to the same package)
- avoid silly pool-related imagery (e.g. like my part scraping and generation script called `poolnoodle.py`).
- maybe it could also make a little sense. In the draft above I was inspired by the lines that you see when you list things, and it also doubles as shooting stars, but maybe for a outside person it is a bit too far fetched.  

In any case I will try to make a few drafts, so we don't have to take the first thing that we see.

carrotIndustries, 2018-06-05:
4th Option that i'll probably go with: Combine pool and project manager into one application that transforms into the proper one depending on what's opened. This also helps to bring pools and projects closer together as needed for #165

closed 2018-06-20 by carrotIndustries

atoav, 2018-06-21:
Very cool, removes a lot of visual and mental clutter and works pretty well!

commit c97b5195 (2018-06-20): unify pool and project manager

fixes #163