horizon-eda/horizon #167: Feature Request: Bevel for Polygons in Footprint Editor

atoav, 2018-05-24:
It is a typical task in the Footprint Editor to have to Bevel edges – as it has to be done for nearly every assembly outline.  

**Status Quo:** Right now to bevel a edge you have multiple ways, each with it's own pitfalls. The right way is to set your grid to a divisor of the horizontal or vertical package width (so the grid points fall onto the outline of your package). Then you click `Add Vertex~ which you then drag to  point where it roughly assembles a 45° angle.

**Problem:** This can be cumbersome, as you have to set and reset the grid and in the end you will not get a perfect 45° angle unless your part is square and aligns with the raster.  

**Solution:** Add a "Bevel Edge" Action where a drag sets the bevel size and the scrollwheel (or some keyboard modifier) sets the stages, similar what blender has:  
[image]  
  
**Alternate solution:** Don't automatically enter move mode after Vertex has been added  
  
**Alternate solution 2:** Allow to set absolute coordinates in right sidebar for Junctions and vertecies

carrotIndustries, 2018-05-24:
The common case of packages with a beveled corner is already handled by the "Draw polygon rectangle" tool using the "decoration" option.

atoav, 2018-05-25:
Ah okay, missed that, cool. This will save some time in the future!  

There are some cases where Bevel would still make sense (e.g. the smd electrolytics I do right now, where you need two bevels), but I think it is not high priority atm.

closed 2018-06-21 by atoav