horizon-eda/horizon #169: Schematic Editor: Cutting and pasting symbols btw. sheets

atoav, 2018-06-22:
Is this possible or am I just overseeing something? Copy/Paste and deletion of the original has the downside that it destroys existing layouts in the board view. I think one should be able to freely move symbols/units between sheets (e.g. if you keep your power pins in one sheet of the powersection).

carrotIndustries, 2018-06-22:
There's the action "Move to other sheet" exactly for this purpose. It automatically grabs anything connected to the symbol you selected so references don't break.

Unfortunately, actions are only listed in the spacebar menu right now, limiting their discoverability. They'll also appear in the context menu in the future.

atoav, 2018-06-23:
Ah okay, didn't realize that. I tried:
- <kbd>Ctrl</kbd>+<kbd>x</kbd> and <kbd>Ctrl</kbd>+<kbd>v</kbd>
- hitting <kbd>m</kbd> and selecting the other sheet while moving stuff  

Do I get one thing right – the Power Unit needs to be on the same page as some other units belonging to the same entity?

carrotIndustries, 2018-07-07:
> Do I get one thing right – the Power Unit needs to be on the same page as some other units belonging to the same entity?

No, they can be placed as desired. The annotation however will use the first occurrence of the component for determining the refdes.

closed 2018-07-07 by carrotIndustries