horizon-eda/horizon #171: Feature Request: Outliner for Schematic Editor

atoav, 2018-06-24:
Maybe in the long term  something like a outliner would be good:  

So for example a list in the left pane below the Sheets list where all parts and named nets are listed. A search field to filter certain parts/nets, a option to look at selected and the ability to select multiple parts at once. Unplaced parts or "ghost parts" as I called them could be grayed out.  

This would also help occations where you want to swap all 100k resistors for 0605 instead of 0805 or so. Just type "100k" select all and right click > Assign Part in the Sheet. In addition with the quite cool live selection between schematic editor and board editor this would make things even more cool – and it is unused screen space after all.

atoav, 2019-01-20:
I gave this another thought. 

I think from a perspective of perception it is just important that every entity that "exists" within the context of a project has to be listed somewhere central. This is crucial for seeing unplaced symbols, and helps you realizing things like avoiding to use 3 different types of 100k resistors.

What I came up with, is sort of a hybrid between BOM and Outliner. This portion of screen space is unused as for now.

[image]  

## In a ideal world
- better overview of the parts used
- ability to select all instances of a part, single instances (by clicking on the Refdes) or all instances of a part on a sheet (by clicking on the sheet number)
- ability to search for parts
- maybe adding "blind" parts, that shouldn't show up on the schematic or the board, but you wanna have in the BOM (this could be free text, for adding some Knobs for a potentiometer, or some nuts and screws for a hole etc.)
- maybe even add some sort of price tag on each part so you can get a feeling for the total cost in real time (although that might be feature creep)

pwnorbitals, 2020-11-15:
Interesting, @carrotIndustries is this still considered ?

carrotIndustries, 2020-11-15:
Yes, see #481