horizon-eda/horizon #176: Tools do not keep state between invocations

endofexclusive, 2018-07-09:
An example:
1. The _Draw line_ tool allows to set the line width. That line width is used for the lifetime of the tool until the user ends the tool. When the user activates the tool again, the line width is reset.
2. The _Draw line rectangle tool_ can change mode between center or corner oriented by pressing the `c` key. The mode is not retained between invocations.

One scenario where retaining the state/mode is useful is when a small non-electrical illustration is added to the schematic. In that case one would want to draw a couple of graphics boxes, lines, arrows and similar. These typically share the same style properties.

However, this issue/feature request is not limited to the line and rectangle tools.

My understanding of the ToolBase class is that the tool object is created when it is invoked and is deleted, together with its state when it ends. It would be nice if tool state state could be retained in some way, possibly per open document. (The state may or may not be saved in the document.)

atoav, 2018-07-11:
Another example where this can be annoying sometimes is placing parts: the initial orientation will always reset when a new part is placed. There can be situations were this is useful (if it is wildly different parts), but more likely it is annoying (e.g. when you place the 4 opamp units of a quad opamp entity and they reset every time).

carrotIndustries, 2018-07-15:
abb4491dcea77f96a23c90598e85906e6d61ece1 adds persistent settings for the draw line and place text. Suggestions for more tools that also need this?

The symbol is a different issue since in that case the tool forgot to apply the last symbol's placement to the new one. 92b63741e95a2fd1ca1b78a80587c12568a3bfed fixed this

Settings get saved globally on a per-editor basis

endofexclusive, 2018-07-15:
This is awesome, thanks!
For line width, it could be useful also for _Draw line rectangle_ and _Draw arc_.

endofexclusive, 2018-07-19:
The line width setting works very well. It is also nice to see the ToolSettings framework in place. Thank you!

pwnorbitals, 2021-08-01:
I imagine this issue can now be closed @carrotIndustries

commit 87f42f9d (2021-08-03): ToolDrawLineRectangle: remember mode

state shared with ToolDrawPolygonRectangle

closes #176
closed 2021-08-03 by carrotIndustries