horizon-eda/horizon #179: Simplifying mask/paste polygon generation for complicated pad stacks

fruchti, 2018-07-22:
It's very nice that Horizon supports arbitrary pad geometries, but making solder mask expansion/paste contraption parametric can be pretty hard. Having `expand-polygon` for padstacks as well with a similar "generate parameter program" button to the courtyard one would be a nice solution IMHO, but maybe you have another solution in mind?

Background: I'm currently trying to build this footprint:
[image]
and I really don't want to calculate all coordinates manually and use `set-polygon`.

carrotIndustries, 2018-07-22:
ae33948912f0a5882357a26134689b87058c1347 makes expand-polygon available for padstacks as well

Mind that your given padstack might/will break the router since the router expects to only get convex pads. IIRC kicad solves this by triangulating the pad, so this shouldn't be too hard to fix

carrotIndustries, 2018-07-23:
5c4ef924b69909cf773fcade8a40068795e50ea2 should enable non-convex pads

fruchti, 2018-07-23:
Perfect! Working exactly as it's supposed to. Thanks a ton!
[image]

closed 2018-07-23 by carrotIndustries