horizon-eda/horizon #185: Bus ripper orientation

endofexclusive, 2018-08-01:
Bus rippers support this style:
[image]
But not this:
[image]

The `BusRipper` class has a `mirror` boolean which currently allows to mirror around the Y axis, but not the X axis. `mirror` could be changed to an `Orientation` instead to support pointing into one of four quadrants. 

Another approach would be to represent the ripper with an angle and length. That could in the long term get away with the 1.25 mm (fixed?) length.

carrotIndustries, 2018-08-01:
While we're at it, should we support bus rippers on horizontal bus lines with the text going vertically? IMO it's not really worth the hassle since we'd end up with 8 orientations total and all sensibly-drawn symbols have their bus ports on the left/right side.

endofexclusive, 2018-08-01:
In my opinion, vertical text in schematics is definitietely a thing to avoid. However, that is just an opinion and there could be situations where other factors can sanction it. I did a web search to try to find authoritative guidelines on schematic text orientation, but could unfortunately not find any. (But lots of opinions on forums etc.)

For Horizon, I would be happy with horizontal text only. If the text becomes to wide, then it could be shortened since it is local to the bus anyway.
                                                                                
If there is going to be a change in the implementation that breaks the file format, like substituting bool mirror into Orientation, then it could be an appropriate time to also add support in the file format for the arbitrary text orientation. Even though the aim is to currently not support it, it would prevent the file format from being a blocker for a future implementation.

closed 2018-08-01 by carrotIndustries

carrotIndustries, 2018-08-01:
File format changes aren't that bad as long as the current version of horizon can open files created with older versions.

I don't think that arbitrary length/angle is something desirable since everything on the schematic is supposed to be aligned to the 1.25mm grid.

endofexclusive, 2018-08-02:
The bus ripper orientation works very well here with 7adfac9. Thank you!

commit 7adfac94 (2018-08-01): more bus ripper orientations

closes #185