horizon-eda/horizon #187: Feature request: multi-line text

endofexclusive, 2018-08-02:
This is a feature request for placing multi-line text in schematic and board editor. A current workaround is to put multiple text fields and manage the line breaks manually.

Related is also text formatting. It would be very cool with direct Markdown support. The CommonMark implementation provides a parser in library form which is independent of the rendering.

endofexclusive, 2018-08-08:
I did some experiments with text formatting.
The result looks like this:
[image]

The implementation is available on my Gitub account in the branch textfmt:
https://github.com/endofexclusive/horizon/commits/textfmt
It is experimental and not ready for pull request (yet). But could work as a discussion point.

Formatting is controlled with in-band format codes in the string and is processed by `draw_text0()`. It allows for bold text and overbar. (`~MCLR` pin in the screen shot). Formating codes can be masked on each call to to `draw_text0()`.

If in-band format codes is the way to go, then the following could be useful: bold, italic, overbar, monospace, newline.

carrotIndustries, 2018-08-08:
Some things to look out for: when text is drawn right-to-left (anchor is on the right side), the string gets reversed. Also make sure the overbar ends up correctly when the text is flipped (i.e. place on bottom layer in board).

TBH, the way text rendering works internally (`draw_text0`) is super messy and hacky and needs a major overhaul anytime soon. 

For example, instead of drawing right away, draw the lines into a buffer first, then draw the buffer. That way, we could get rid of reversing the string before drawing.

The text drawing code has been adapted from OpenCV: https://github.com/opencv/opencv/blob/9787ab598b6609a6ca6652a12441d741cb15f695/modules/imgproc/src/drawing.cpp#L2262
In that process a lot of features such as different fonts went overboard. It'd be nice to re-add these.

carrotIndustries, 2018-11-17:
317cea68e0b1f1d6fa37006db0509b097d77ba48 adds multiline text

atoav, 2020-02-23:
This can be closed right?

carrotIndustries, 2020-02-23:
We have multiline text but still no inline formatting. Since the fork containing the inline formatting seems to be gone, I'll close this issue. If someone wants to add inline formatting for bold/overline they should open a PR.

closed 2020-02-23 by carrotIndustries