horizon-eda/horizon #190: Schematic editor: no connect symbol

endofexclusive, 2018-08-06:
I can not find out how to mark a symbol pin as "no connect" in the schematic editor.

It can look like this in Orcad Capture:
[image]
and Altium:
[image]

It could possibly be "emulated" in Horizon with a dedicated part which has an empty package and/or no pad->pin mapping. However, that probably requires a short net segment between the symbol pins, and does not convey "no connect" information to the rule checker.

carrotIndustries, 2018-08-07:
Use the "Set all unconnected pins NC" tool on a symbol to make all unconnected pins NC pins. To change individual pins use the "Set pins NC" tool.

closed 2018-08-07 by carrotIndustries

endofexclusive, 2018-08-07:
It works perfectly. Thanks!
Added the key combination "p n" on my setup.

commit 12f30bcc (2018-08-07): support NC pins

closes #190