horizon-eda/horizon #192: Call for installer artwork

carrotIndustries, 2018-08-07:
Now that we have windows installer WIP (see `wix` branch), we need some artwork to make it pretty: https://github.com/carrotIndustries/horizon/tree/wix/wix/Bitmaps

Dialog.bmp will be visible on the first and last page of the installer like this:
[image]

banner.bmp will be visible in the top bar
[image]

Anyone got some ideas what to put in there? I thought about putting the application icon in the banner and some schematic/board screenshot in left right part of the dialog. @atoav?

atoav, 2018-08-08:
I am on it, you will get some drafts by tomorrow evening

atoav, 2018-08-08:
Maybe something among these lines?  

[image]

carrotIndustries, 2018-08-08:
Looks great! Is the pixely look supposed to part of the aesthetic? What are your ideas on applying your concept to the installer images?

atoav, 2018-08-09:
The pixelation is actually due to the shitty screenshot I used as a source here. The cleaner solution would be to actually take the pdf/vector files and recolor it in accordance to the color scheme.  

That might not be necessary, because the thing I posted here is 4 times the resolution that bitmap has – maybe bicubic downscaling already helps a lot.  

About the concept of the installer image, I imagined something among these lines:  
[image]

For the banner I would just use the horizon logo on a neutral background, like the inkscape folks did it. I can certainly prepare that also.

carrotIndustries, 2019-05-12:
> One thing: one of my hard drives broke so the source files for that splash specifically are lost because I didn't have them in the backup for some reason. I could redo another one, or we could use that one, however you like. Redoing another one could make sense if we want higher resolution, but I am not sure how close I'd get to the same thing again.

I guess no one will mind if the new version looks slightly different, so it'd be great if you could redo them. 

To make sure we don't lose any files anymore, we now have https://github.com/horizon-eda/horizon-artwork

You might also want to move https://github.com/atoav/horizon-logo over there.

atoav, 2019-05-12:
>  it'd be great if you could redo them  

I sure can. Might just take till the end of the week or so

> You might also want to move https://github.com/atoav/horizon-logo over there.

Done

atoav, 2019-05-13:
I designed two variants, check out [https://github.com/horizon-eda/horizon-artwork](https://github.com/horizon-eda/horizon-artwork).

carrotIndustries, 2019-05-13:
Looks cool, for my part, I prefer the left one.

We also need to decide on a less generic and more inspiring tagline than "a free EDA package". That way, we'd also get rid of having EDA in there twice.

atoav, 2019-05-13:
I agree, I felt the same, this is a case of unnecessary redunancy, and if you don't know what "EDA" means precisely it might make you feel a little silly. But I didn't had any better idea so far. I mean "free" is good as a general idea. Maybe something about electronics, circuit design etc?

fruchti, 2019-05-14:
I love the artwork! Modern and clean, just like Horizon ;)

> We also need to decide on a less generic and more inspiring tagline than "a free EDA package". That way, we'd also get rid of having EDA in there twice.

I agree, that struck me as odd at first, too. Though I like the current tag line for its humbleness and “free” having various positive meanings. An idea would be to focus on the legacy-free concept? Something along the lines of “a modern experience” comes to my mind (but that is a bit generic).

Nitpick @atoav: Is it just me or should the text in the logo be a smidgeon lower to be (optically) centred? Maybe I've just been staring at it for too long.

atoav, 2019-05-14:
> Is it just me or should the text in the logo be a smidgeon lower to be (optically) centred?   

Legitimate critique, I already had it on the list myself : )

carrotIndustries, 2019-05-14:
I've thought about something similar to "next-generation open source electronic design", but at second though that's probably too long and too pretentious. Perhaps just "free electronic design automation"?

atoav, 2019-05-18:
I researched a little to figure out what others do:

| package | tagline |
| --- | --- |
| LibrePCB | _A new, powerful and intuitive EDA tool for everyone._ |
| KiCAD | _A Cross Platform and Open Source Electronics Design Automation Suite_ |
| Altium | _Easy, Modern and Powerful PCB Design Software_ |
| Eagle | _PCB design made easy_ |
| Fritzing | _Electronics Made Easy_ |
| Proteus | _PCB Design Software and Embedded Simulation Tools for modern EDA development._ |

Let's brainstorm a little (just a collection of words)
- free, open, open source, libre, personal
- easy, intuitive, direct, clear, straight forward
- package, solution, software, tool, suite
- modern, next-generation, fresh, clean, state-of-the-art
- EDA, ECAD, electronic design automation, circuit design tool, circuit design package, ...

To be expanded: 
| text | pro | con |
| ---- | ---- | ---- |
| _a free EDA package_ | <ul><li>short</li><li>humble by use of "a" (indicating there are other solutions)</li><li>_package_ indicates a working solution from start to finish</li><li>_free_ makes clear what it is</li><ul> | <ul><li>EDA is redundant with the name of horizon-eda</ul></li> |

The more I think about it, the more I like our current slogan, despite the recursive nature of the EDA. I think we should only change it if we come up with something substancially better.

carrotIndustries, 2019-05-18:
any opinions on "free electronic design automation"?
Pro: 
 - short
 - explains what EDA is
 - not pretentious

Con:
 - generic, EDA also includes chip and FPGA design

atoav, 2019-05-18:
> any opinions on "free electronic design automation"?

Sure. My feeling was that it is a bit too "unwieldy". It explains EDA, which isn't bad, but to be honest the word "automation" feels a little bit out of place. IMO it doesn't adequatly descripe what makes horizon feel like it does and is a little bit too generic for my taste.

I wondered if we should focus on the problem this software aims to solve for the user, instead of describing precisely what it does: 
-  _"a free circuit design tool"_
- _"a free PCB design package"_
- _"a free schematic and circuit creation package"_

This feels a little like writing a Haiku: it should be short and every word counts, but in the end it should feel totally effortless.

carrotIndustries, 2019-05-21:
I've had some more thoughts about this. Does the tagline really need to include something along the lines of too/app/package?

How about "free schematic and PCB design"?

atoav, 2019-05-23:
> Does the tagline really need to include something along the lines of too/app/package?

No of course not, we are completely free here and we should explore other options as well. The good thing after all is that the existing tagline isn't totally wrong, so we don't have to take anything that doesn't convince us.

> How about "free schematic and PCB design"?

I understand the direction you want to go in with this, but I think this would strictly speaking refer to:
- the act of freely designing schematics and PCBs
- some kind of service that designs circuits and schematics for you – free of charge

While I like the former, I am a little worried about the latter. The ambiguity of the word "free" with it's multiple meanings must always be considered, and it should work for and not against us:

If we take for example _"free schematic and PCB design**er**",_ both meanings of "free" would refer to the "designer", and while this could also be refering to a person, it would be very clear from the context that it is not. But then we would be back to mentioning the package again..

carrotIndustries, 2019-05-24:
> While I like the former, I am a little worried about the latter. The ambiguity of the word "free" with it's multiple meanings must always be considered, and it should work for and not against us:

I guess you're right on that one.

> I think we should only change it if we come up with something substancially better.

Seems like a good idea, otherwise we'll never get anything done. If we come up with anything better in the future, it's not a big deal to change.

carrotIndustries, 2019-08-30:
There's a bit of an issue with the application of the new artwork to the installer:

[image]

Judging from the inkscape installer, the artwork on the side should be about 164x312 px. The height is fixed, a few pixels less on the width aren't too bad.

atoav, 2019-08-30:
Hey I added another splash with the correct proportions here: https://github.com/horizon-eda/horizon-artwork/tree/master/splash/PNG

carrotIndustries, 2020-10-08:
Closing since we now have installer artwork.

closed 2020-10-08 by carrotIndustries