horizon-eda/horizon #193: Padstack parameter issues in UFQFPN20

fruchti, 2018-09-12:
Pad width and height parameters for SMD rectangular pads are no longer accepted in the UFQFPN20 package.
[image]
The problem persists even if all SMD rectangular pads are removed from the footprint and it is reloaded.

This seems to be the only affected package but as it hasn't been changed since its initial commit, I'd consider this a Horizon bug rather than a pool issue.

fruchti, 2018-09-12:
`git bisect` tells me the problem exists since ae33948912f0a5882357a26134689b87058c1347.

carrotIndustries, 2018-09-12:
Looks like we've had two incompatible implementations of `set-polygon` before that commit:
 - `Package::MyParameterProgram::set_polygon` that makes a polygon a predefined shape (only rectangle for now)
 - `Padstack::MyParameterProgram::set_polygon` that sets the vertices based on the arguments placed on the stack

Me trying to be smart back then saw two identically-named commands and picked one without much thought.

Since both commands have their right to exist and we should strive to have identical commands available across all items, what are your thoughts on cleaning up the mess I made?

Since your padstack is the only one using `Padstack::MyParameterProgram::set_polygon`, renaming that `set-polygon` to `set-polygon-vertices` appears to be best solution to me.

fruchti, 2018-09-12:
Now that you mention it, I see that the SMD rectangular pads being misadjusted were actually a red herring. I guess after one bevelled pad's parameter program fails, Horizon gives up on those?

Anyhow, that's partly my fault, too, as I introduced the second notion of `set_polygon` in #41 exactly for those pads. `set-polygon-vertices` is actually a clearer name and I'm all for that solution.

carrotIndustries, 2018-09-14:
> I guess after one bevelled pad's parameter program fails, Horizon gives up on those?

yes, 7fdda32a42b62ba56a76768daad0b4a8e2fcb87f tries to improve error reporting in that regard

> set-polygon-vertices is actually a clearer name and I'm all for that solution.

done in 2fa114cec3ea719791949dc8e5808735282e2d64

closed 2018-09-14 by carrotIndustries

fruchti, 2018-09-15:
Thanks! [This PR](https://github.com/carrotIndustries/horizon-pool/pull/58) makes the package look like it should again.