horizon-eda/horizon #194: Usefulness of “keep horizontal” for pin names

fruchti, 2018-09-13:
It does exactly what its name says, but I can't really think of a context where this behaviour would be desirable.
[image]
Under normal circumstances it looks bad in 3/4 cases. Maybe a “rotate pin name by 90 degrees” option is better here?

carrotIndustries, 2018-09-14:
a43aeed62d9b786c7cfab4b201635f7a4b5f59bd fixes the text placement

The idea behind this feature is to avoid vertical text in all orientations.

fruchti, 2018-09-15:
Definitely looking better! Yet I still don't really know when this would be nice to have. Did you have an example in mind for this feature?

Would splitting this configuration into two options like

* name perpendicular to pin and
* keep pin name orientation when rotating symbol

be an option?

carrotIndustries, 2018-09-18:
This feature came to be since vertical text is something i'd like to avoid wherever possible, hence the option to force the text so stay always horizontal.

Are there any use cases where that'd be undesired?

fruchti, 2018-09-18:
I'm absolutely with you in that vertical text should be avoided. The problem here is the rotation of the labels then changes relative to the symbol, so the pin generally needs more space inside the “symbol body”. The symbol pictured above would need to be larger even with the fixed pin name placement (`VDD` collides with the diode with this size). In other words, even when it doesn't help in 99% of the cases, I'd have to make the symbol larger for the 1% we almost never see.

Imagine a microcontroller symbol with pins mostly on the left and right sides with a few power supply pins on the top and bottom (assuming we decide on this layout). The standard orientation will be used in pretty much every case. Having the power supply pins' names horizontal too would look nice and also save space in most cases. If you rotate the part, though, it'd get difficult: You wouldn't keep all pins horizontal because that would require tremendous space around each pin so the names wouldn't collide. Yet the power supply pin names would change orientation and require that space, voiding one of the advantages.

I try to keep my symbols as compact as possible while keeping them visually structured. This feature in its current form doesn't really help me there and a think a good compromise between keeping pin name horizontal all the time and reasonable symbol compactness would be achievable pretty easily.

fruchti, 2018-09-20:
Looks good, thanks!

closed 2019-02-06 by fruchti