horizon-eda/horizon #196: Zoom steps are too far apart

atoav, 2018-10-11:
Somewhere inbetween the last time I have been active and the latest set of builds the zoom behaviour changed for me. Before it felt "normal", right now the Zoom steps are unpractically far apart, for some reason. Or is this intentional?

I tried to change the smooth zoom settings with no effect (other than removing the interpolated animation inbetween zoom steps).  

A screencap says more than a thousand words:  
[image]  

I run the latest horizon build 5d6b2dd7 on a x64 Windows 10 machine.

carrotIndustries, 2018-10-11:
Can you post a screenshot before and after zooming so we can work out the scale factor?

atoav, 2018-10-11:
Sure! This is one step:  
Before:  
[image]  
After zooming one step:  
[image]  

The A4 schematic fits on the screen in the zoomed out level and is a bit to close IMO. Instead of adjsuting the position of the zoom steps, I think a finer raster would be cool. Maybe this could be a setting in `preferences → zoom`?  

We could use this as a default and then just add a list of fractions that defines the size of intermediate steps, e.g.: 1, 1/2, 1/4, 1/8, 1/16 etc. So if somebody likes smaller steps, they can do that.

carrotIndustries, 2018-10-11:
Ok, that's definitely a bug. Normally, the zoom factor should be 1.5, but in your case, it's 3.37 which is roughly equivalent to 1.5³. So it looks like you're scrolling 3 clicks at a time.

atoav, 2018-10-11:
I checked my mouse settings and the math checks out. Windows was set to scroll 3 lines at a time.

carrotIndustries, 2018-10-11:
I've triggered a build of the commit before the smooth scroll change: https://ci.appveyor.com/project/carrotIndustries/horizon/builds/19441710

If the issue is still present in that build, we've got a gtk bug/regression.

atoav, 2018-10-11:
Ok damn, I cannot test this, that build has a missing dll : )

carrotIndustries, 2018-10-11:
Try copying the DLLs from the latest build to that build.

atoav, 2018-10-11:
Okay nice – this worked.

The results:  the step size in the old build is influenced by the windows scroll settings in the same way as in the new build.

carrotIndustries, 2018-10-11:
There've been some changes to scrolling in gtk on win32: https://gitlab.gnome.org/GNOME/gtk/commits/gtk-3-24/gdk/win32 looks like I'll need to dig out my windows laptop to investigate this further...

atoav, 2018-11-18:
I will leave this open for now, just as a reminder

commit c8d71e12 (2019-01-11): Revert "disable smooth scrolling on win32"

This reverts commit 3e089c1520960b257886e1ec1e42575e6f597dfd.

has been fixed in gtk

fixes #196
closed 2019-01-11 by carrotIndustries