horizon-eda/horizon #199: Flickering Tooltips in Pool editor

atoav, 2018-10-24:
See gif below, the capture is a bit slow, this is faster. I forgot to turn on capturing for the pointer:  
[image]

carrotIndustries, 2018-10-24:
I was able to reproduce this error on windows, but have't yet figured out whether this is a gtk bug or I'm just using gtk incorrectly.

carrotIndustries, 2018-12-09:
corresponding gtk issue https://gitlab.gnome.org/GNOME/gtk/issues/1504

carrotIndustries, 2018-12-11:
Seems like this issue has already been fixed in gtk (see link above), so the only thing left to do is wait for the next point release.

carrotIndustries, 2019-01-11:
gtk release is out, is included in latest builds