horizon-eda/horizon #200: Adding a 3D model crashes package editor if more than one package editor opened

atoav, 2018-10-24:
Adding a 3D model crashes package editor if more than one package editor opened. See error messages below:  
[image]
[image]

For the sake of stability I suggest that a crash in the 3D editor should not drag the package editor with them, if possible. Maybe it makes sense to only allow one package editor window to be opened at once?

carrotIndustries, 2018-10-24:
opengl error 1285 is GL_OUT_OF_MEMORY, so I guess your 3D Model must be quite large or your GPU memory particularly small. Making OpenGL errors non-fatal definitely is something to consider to improve stability.

atoav, 2018-10-24:
I accidentally had two package editor windows open – weirdly enough the _very_ same 3D file loads without any trouble (and without taking time) if I close one of them. That is why I suspect there must be some kind of other source of the excessive memory usage (also, I work as a VFX freelancer, that means I know the memory limits of my GPU quite well).  

I managed to narrow it down: in order to produce this crash the second (inactive) window needs to hold unsaved changes. That means:  
1. Take a part and edit it (Leave the window open)
2. Duplicate that part and edit it
3. Add a 3D model to the duplicated part
4. Crash

carrotIndustries, 2018-10-24:
Since the two package editors are running as independent processes one editor shouldn't affect the other in any way. There's a build in the queue that has some more error checking built in: https://ci.appveyor.com/project/carrotIndustries/horizon/builds/19766965 Give that one a try once it's done.

atoav, 2018-10-24:
Okay, I will as soon as I have time