horizon-eda/horizon #201: Question: Parametric Data for Parts

atoav, 2018-10-24:
I have seen parametric data on multiple parts and because I am in the process of adding multiple parts – I wondered about the details..    

In one of the CPL caps .json it says:  
```
"parametric": {
        "table": "capacitors",
        "value": "2.200000e-11",
        "wvdc": "50.0",
        "characteristic": "C0G/NP0"
    },
```  

It is clear these fields are make sense for every cap, but are there more keys that I can set? E.g. wvac, the lead spacing, part height or similar?  
Are all keys allowed or is there a set of allowed keys? If not, is there a list of allowed keys?  
What about the value for `table` – is there a list of valid tables?  

No stress or anything I was just wondering, maybe this would be something for the wiki..

carrotIndustries, 2018-10-24:
Pretend that this feature is not there. For now parametric data isn't used for anything and is only implemented since it seemed easy back then.

Some ideas where parametric data might be used in the future:

- parametric search in the part browser
- displayed in schematic to indicate voltage rating of a capacitor and the like

atoav, 2018-10-24:
Okay, I will leave that stuff out for now. I think a well implemented parametric search would be at the top of my whishlist, but these things need time and should not be taken too lightly.  

I must say I am very impressed with the progress and shape of the project, thanks again for the persistence : )

closed 2018-10-24 by atoav