horizon-eda/horizon #202: Duplicate Part crashes whole application if directory doesn't exist

atoav, 2018-10-24:
The Pool Editor's `Duplicate Part` Function crashes the whole application if the directory specified in the new path doesn't exist.  

### Expected Behaviour
Create missing paths (like `mkdir -p` in bash followed by a `cp`)  

### Observed Behaviour
Whole Application crashes without warning  

### Steps for Reproduction
1. (In the Pool Editor) Select a part and click `Duplicate`
2. Just duplicate the part and enter a path pointing to a nonexisting (but writeable) directory
3. Crash

carrotIndustries, 2018-10-24:
1d8e136e9bc5b61db6c976cac4bd58faaf2ae68c at least fixes the crashing part of this issue

atoav, 2018-10-24:
I used this thing for the past hour so two minor usability nitpicks (feel free to ignore them):  
- It would be cool to have directory and filename seperated. Especially long paths become unwieldy and often you just need to change some small value of the filename
- Would be cool if editing the strings in these field actually had a <kbd>Ctrl<\kbd> + <kbd>Z</kbd> history to undo fuckups like overwriting paths that you didn't want to overwrite etc.  

Dunno how hard these are to implement, but it is not too important.
(Good night and thanks for the quick responses)

closed 2018-10-31 by atoav