horizon-eda/horizon #203: [UI] Change net label default size from 1.5 to 1.3

atoav, 2018-11-08:
I suggest a change of the default sizes of netlabels from 1.5 to something smaller because of how they look with the default spaced pins: 

1.5  
[image]  

1.3  
[image]  

1.329 (this is roughly where the lines of the labels line up)  
[image]

Of course this will always be a compromise between font size and grapcial looks – maybe the font could be bigger and the label margins a tiny bit smaller, etc.  I think the default solution should be something that works well on commonly used parts without a need for manual adjustments that is why I think non-overlapping labels would be nice : )

atoav, 2018-11-08:
Closed, I am an Idiot, this seems to work as soon as you name the nets for some reason?

atoav, 2018-11-08:
Okay maybe I am not an idiot, it seems that certain characters have a higher line height than others:  
n the case of "? plz fix" the questionmark is higher than the rest. In the case of the offsheet references it is the square brackets.  

You can see here (all with 1.5mm height) how the first two labels are well spaced, while the last 3 labels all have different heights because the square brackets are higher than the questionmark which is in turn higher than a lowercase label composed of dots:  

[image]

Maybe this is intentional by design, but I'd suggest to look into keeping it at the same height at least for all regular labels

carrotIndustries, 2018-11-14:
971b2a4c421c66346b044e4fa1c5fa82366953ca persists net label size in tools

atoav, 2018-11-18:
Looks better now : )

closed 2018-11-18 by atoav