horizon-eda/horizon #206: [UI] Allow fine grid movement with Alt + Arrow

atoav, 2018-11-18:
In #154 I proposed the general arrow key movement of parts, which I happen to use extensively since then (probably 50% of all movements I do). A minor detail that I also proposed back then was to allow the use of the same Alt-Modifier you implemented for the mouse move command <kbd>m</kbd>.  

The idea was to use <kbd>Alt</kbd> + <kbd>←/→/↑/↓</kbd> to move things on the finer grid via keyboard.  

This is obviously not very high priority, but I'd still love to see this, because keyboard movement especially shines when you have to move stuff just a tiny bit.

carrotIndustries, 2018-12-03:
Would starting the move tool via Alt-arrow resulting in all subsequent arrow keys being interpreted as fine movements work for you?

atoav, 2018-12-04:
To be honest I'd like it more if this would be global in a similar fashion like <kbd>←</kbd>, <kbd>↑</kbd>, <kbd>→</kbd> and <kbd>↓</kbd> are at the moment. That means while <kbd>Alt</kbd> is beeing pressed every arrow key would move on a fine grid (just like it moves things on the coarse grid without <kbd>Alt</kbd>). This way you can alternate quicker between fine and coarse grid movement, which is IMO more intuitive.

But I suspect this would be harder to implement than your proposed solution? If this is a problem, then just go ahead and implement it the way you described.

atoav, 2019-02-04:
So I am not the only one : )

@frmdstryr You know you can use math operations in the grid field too? e.g. if you type `/10` after the existing value he should divide by ten and if you type `*10` he should multiply by ten

carrotIndustries, 2019-02-04:
The reason why this has been stalled for so long is that I haven't yet figured out a satisfactory way to implement this. Tools/actions are activated by the key sequences defined by the user, adding move fine left/right/up/down would require the user to change the key sequences in sync with the fine grid modifier to maintain consistency. We'll also need to figure out whether to perform a fine move operation in the tool itself. Right now, a tool doesn't really know about preferences.

atoav, 2019-02-04:
I am not sure about this, but what if instead of changing the tool, <kbd>Alt</kbd> would change the environment, unless used in a combination specified in the users preferences. 

That means pressing <kbd>Alt</kbd> could switch the grid from coarse to fine, while the arrow keys just Move by one grid unit, whatever it may be at the moment.

Where this could get a bit messy is with all Shortcuts that rely on Alt as a modifier key, but I think switching the grid momentarily is not that much of an impact.

But maybe this makes no sense at all from a code perspective..

frmdstryr, 2019-02-04:
Could #230 be reviewed to see if it's an acceptable solution (or at least in the right general direction)?

commit 3b12b58e (2019-02-08): add fine move

fixes #206
closed 2019-02-08 by carrotIndustries