horizon-eda/horizon #212: Pool manager: Replace line breaks from input fields

atoav, 2018-12-03:
While creating parts I often paste text (e.g. a MPN) into the according field. These fields will _not_ display line breaks, but they will be shown in the parts list later. This means there are three possible outcomes:
1. I just don't realize that there was a line break
2. I have to retype the thing
3. I have to paste it into some other field that replaces `\n` and copy it again

All of these are not really to be desired.

## Solution
Either:
* Replace line breaks in the fields
* Allow line breaks, but display them

carrotIndustries, 2018-12-03:
Ok, on Linux these show up as boxes
[image]

Gtk offers a simple option to work around this by truncating everything after the first `\n`.
Alternatively, writing some custom code that hooks into the paste event might be the way to go.

atoav, 2018-12-04:
Ah okay, now I get the rationale. On Windows they are displayed as zero space ehich essentially makes them invisible in the text field but visible in the list.

Best from a usability perspective would be to just replace “\n” with space.

carrotIndustries, 2018-12-04:
So I've had a a chat with Gtk folks on their IRC channel: There's no sensible way to intercept paste data on it's way to the entry, the only workable workable solution on the entry level would be to truncate everything after the first newline.

How about something like this instead? The black box is a tooltip that appears after hovering over the icon.
[image]

commit 4e936f45 (2018-12-08): sanitize entries in pool manager

fix #212
closed 2018-12-08 by carrotIndustries