horizon-eda/horizon #214: Feature Request: Better Parts search

atoav, 2018-12-04:
## Problem
Searching for MPN works only with exact strings, which only works well for _some_ MPNs. Certain MPNs or ranges thereof become hard to search if the MPN e.g. contains a variable part in the middle that you would like to ignore. 

The reason I even come up with this is, because _the pain is real_. With [PR #48 on the horizon-pool](https://github.com/carrotIndustries/horizon-pool/pull/48) I added over 6000 resistors to my library. Filtering something simple like a 0805 10k resistor proved harder than it should (see footnote ¹ for screenshot). This way it can take up to **2 minutes (!)** till I find the fitting part, depending on the time of the day and whether I had a coffee yet. 

Tags and manufacturer already work _great_ to narrow down searches. But a real parametric search will probably not be around anytime soon (and you can't always rely on accurate parametric data anyways). That is why I think it is really important to have a capable text search.

## A possible solution
If I checked right the [lines 77 and onwards in pool_browser_part.cpp](https://github.com/carrotIndustries/horizon/blob/master/src/widgets/pool_browser_part.cpp#L77) do the magic. I suspect under the hood the SQL query "parts.package WHERE parts.MPN LIKE ? AND parts.manufacturer " just queries the database for Parts containing the search input.  

My suggested solution would be to also split the search input on each space and then query for each of the parts with a `AND` inbetween. Spaces are rare in MPNs anyway and that way we could maybe get close to a behaviour like displayed here:  

[image]  

↑ I am searching a shortish BOM for multiple values with [skim](https://github.com/lotabout/skim) (a Rust port of fzf). As you can see:
- a search input of `CPL 50V` yields the CPL cap and it ignored the part in the middle
- a search input of `0805 100K` yields the right 100k resistor without it having me to type out the complete MPN out of my head

If I would have anything vaguely resembling skill when it comes to C++ I would gladly implement it myself, but I am a Rust guy..

----------
¹ See:  
[image]

carrotIndustries, 2018-12-04:
So basically, you need wildcards, right? Since the search inputs are used in a `LIKE` query, `%` is the wildcard character.

> Tags and manufacturer already work great to narrow down searches. But a real parametric search will probably not be around anytime soon (and you can't always rely on accurate parametric data anyways). That is why I think it is really important to have a capable text search.

Parametric search might land in the near future since searching by value by guessing the encoding the manufacturer happened to use really isn't the way to go.

I've also though about introducing a categories since these are easier to discover than tags. Your thoughts?

atoav, 2018-12-09:
## Tags
To be honest, I think the system as it is now (hard to discover user tags), _does_ work in some way for a certain set of problems. It just doesn't work for fine grained parametric properties, like listing all 0.1% resistors with 0805 package and a value of 2k.  

Tags are an easy but powerful system, but inherently unhierarchic. And their biggest problem is that they can get convoluted if tag discovery isn't working well.  

I would see them as a more free floating addition to categories (whoose precise nature needs some discussion I think).  

## Categories
The crucial question to ask here is: What problem are we trying to solve with categories?  

Some suggestions:
- allow "browsing" for parts without any specific goal in mind
- define custom behaviour and data fields per category (parametric data..)
- get some semantic relationship into the thing (E.g. you can search `transistor` to find `NPN` and `PNP` and then limit the scope of a search more. Additionally a `SOIC-8` package is a child both of `ic` and `smd`)

Remaining questions:
- can parts have multiple categories at once? I think this can make sense
- how would we deal with the multiple levels on a hierarchy and the available fields? inheritance?
- what exactly does a category change? The search GUI? available fields uppon parts definition? More?
- will the hierarchy be created before and then applied to all parts, or will we create the hierarchy "on the go" as needed, and then from time to time step in and clean things up?
- how will we deal with parts that are not assigned or have no parametric data (especially when it comes to search)?

I have the feeling that we should avoid getting a rigid hierarchical (tree) structure, but rather a node-like structure, because some things obviously could belong to multiple categories at once and the top goal is increasing discoverability and not a definite taxonomy of the world of electronic parts. 

I can see Categories work quite well for packages as well. For Units/Symbols/Entities it is a bit harder, but it would probably be a more generic version of the Parts thing.

carrotIndustries, 2018-12-23:
ef8ece800c5833bea6c45c75ce7d384573e56e9e adds an improved tag entry widget

having played around with this during development made me reconsider needing categories since the new tag widget makes tags easily discoverable

atoav, 2018-12-23:
Hey that works quite well, thanks a lot!

There might be some minor usability improvements we could think about (e.g. binding <kbd>Tab</kbd> to that + button)

carrotIndustries, 2018-12-23:
> It is a bit slow still (e.g. the time between pressing Enter and a new field opening is ~2s on my machine wiht my pool)

Does this appear to depend on pool size? Is typing in the search entry slow as well? There's quite a lot of SQL going on in the search popup.

atoav, 2018-12-23:
> Does this appear to depend on pool size? 
It appears to be directly connected to the amount of filtered parts (aka the length of the returned list). E.g. when I type `resistor` (6930 parts) it is really slow, adding `th` (1866 parts) is already considerably faster and then adding `vertical` (933 parts) is almost acceptable.  

I again have no idea which beasts you are wrangling here with Gtk (never used it), but the way I'd approach this is probably using some sort of multithreading so the input is responsive, while some other thread does the heavy lifting. If you also limit the initial query results to the first 50 (more can't be displayed at once anyways) and load the rest after the initial results are displayed it could become really fluid. 

So the idea basically is that the the gui code does only need to worry about its own stuff (evaluating inputs, displaying some list of parts), while another thread continuosly spits out more and more results.

Don't know how familiar you are with multithreading (or real asynchonous stuff) tho...

carrotIndustries, 2018-12-24:
Can you point me to the branch with that many parts? I tried your yageo branch with ~3k parts and still found it to be usable. Is regular searching slow as well? I think we might be on something similar to #161

atoav, 2018-12-24:
> Can you point me to the branch with that many parts?

Sure: I have all my branches combined with master at [a branch called _work_](https://github.com/atoav/horizon-pool/tree/work). This is essentially my pool I use for stuff.

Regular search is blazingly fast and seems to take no impact from the tags. This just affects tags as far as I can judge it. Keep in mind that I am on a windows machine as well, maybe this could also make a difference..

carrotIndustries, 2019-01-02:
As far as I can tell this issue should have been there before the introduction of the new tag search. Can you confirm this?

atoav, 2019-01-03:
My issue was: when adding tags with many parts the GUI (in that case the adding of a new tag) has noticeable lag (which I assume to become only worse with higher numbers of parts). I did not encounter any lag with the Manufacturer or MPN search fields in any way (e.g. me having to wait for my Input to show up on screen).  

So it _might_ have been a issue all along, but I never really saw it happening since you fixed #161 with 914e854