horizon-eda/horizon #215: Included Pool's 3D-Models won't show up

atoav, 2018-12-05:
The 3D-model in question shows up as expected when I edit the included pool directly. As soon as I change to the _including_ pool, the 3D model won't show up. This obviously also affects board views that make usage of included parts.

carrotIndustries, 2018-12-09:
Surprisingly these two issues are largely independent:

870841dea9b9bb90cc834f6e67e01e46eef292f9 fixed the 3D models not showing up in the package editor

for the board editor, the culprit is the 3D model cache. We've got two options:
 - Copy all models from all pools to the cache/3d_models directory, might lead to conflicts but keeps things easy
 - Create a subdirectory for each pool

What are your thoughts on that?

carrotIndustries, 2018-12-11:
So what do we want?
Before I read your comment, my idea was for each package to load the 3D models from the pool it originates from.

Your suggestion of 3D models with the same path overriding 3D models in the parent pool probably is the more consistent solution with respect to how all the other items in the pool are handled.

atoav, 2018-12-11:
> Before I read your comment, my idea was for each package to load the 3D models from the pool it originates from.

I didn't know this was possible, but it makes totally sense. I'd go for this now that you mentioned it

commit eb9d7472 (2019-01-01): fix 3d models for included pools

should fix #215
closed 2019-01-01 by carrotIndustries