horizon-eda/horizon #216: make: *** Keine Regel vorhanden, um das Ziel „.git/HEAD“

artbody, 2018-12-07:
unter Gentoo Linux

```
horizon-master # make
c++ -c -Isrc -I3rd_party -g3 -D_USE_MATH_DEFINES -DGLM_ENABLE_EXPERIMENTAL -fdata-sections -ffunction-sections -pthread -I/usr/include/yaml-cpp -I/usr/include/gtkmm-3.0 -I/usr/lib64/gtkmm-3.0/include -I/usr/include/atkmm-1.6 -I/usr/include/gtk-3.0/unix-print -I/usr/include/gdkmm-3.0 -I/usr/lib64/gdkmm-3.0/include -I/usr/include/giomm-2.4 -I/usr/lib64/giomm-2.4/include -I/usr/include/pangomm-1.4 -I/usr/lib64/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib64/glibmm-2.4/include -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/cairomm-1.0 -I/usr/lib64/cairomm-1.0/include -I/usr/include/sigc++-2.0 -I/usr/lib64/sigc++-2.0/include -I/usr/include/cairo -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libdrm -I/usr/include/libpng16 -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/util/uuid.cpp -o src/util/uuid.o
c++ -c -Isrc -I3rd_party -g3 -D_USE_MATH_DEFINES -DGLM_ENABLE_EXPERIMENTAL -fdata-sections -ffunction-sections -pthread -I/usr/include/yaml-cpp -I/usr/include/gtkmm-3.0 -I/usr/lib64/gtkmm-3.0/include -I/usr/include/atkmm-1.6 -I/usr/include/gtk-3.0/unix-print -I/usr/include/gdkmm-3.0 -I/usr/lib64/gdkmm-3.0/include -I/usr/include/giomm-2.4 -I/usr/lib64/giomm-2.4/include -I/usr/include/pangomm-1.4 -I/usr/lib64/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib64/glibmm-2.4/include -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/cairomm-1.0 -I/usr/lib64/cairomm-1.0/include -I/usr/include/sigc++-2.0 -I/usr/lib64/sigc++-2.0/include -I/usr/include/cairo -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libdrm -I/usr/include/libpng16 -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/util/uuid_path.cpp -o src/util/uuid_path.o
c++ -c -Isrc -I3rd_party -g3 -D_USE_MATH_DEFINES -DGLM_ENABLE_EXPERIMENTAL -fdata-sections -ffunction-sections -pthread -I/usr/include/yaml-cpp -I/usr/include/gtkmm-3.0 -I/usr/lib64/gtkmm-3.0/include -I/usr/include/atkmm-1.6 -I/usr/include/gtk-3.0/unix-print -I/usr/include/gdkmm-3.0 -I/usr/lib64/gdkmm-3.0/include -I/usr/include/giomm-2.4 -I/usr/lib64/giomm-2.4/include -I/usr/include/pangomm-1.4 -I/usr/lib64/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib64/glibmm-2.4/include -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/cairomm-1.0 -I/usr/lib64/cairomm-1.0/include -I/usr/include/sigc++-2.0 -I/usr/lib64/sigc++-2.0/include -I/usr/include/cairo -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libdrm -I/usr/include/libpng16 -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/util/sqlite.cpp -o src/util/sqlite.o
c++ -c -Isrc -I3rd_party -g3 -D_USE_MATH_DEFINES -DGLM_ENABLE_EXPERIMENTAL -fdata-sections -ffunction-sections -pthread -I/usr/include/yaml-cpp -I/usr/include/gtkmm-3.0 -I/usr/lib64/gtkmm-3.0/include -I/usr/include/atkmm-1.6 -I/usr/include/gtk-3.0/unix-print -I/usr/include/gdkmm-3.0 -I/usr/lib64/gdkmm-3.0/include -I/usr/include/giomm-2.4 -I/usr/lib64/giomm-2.4/include -I/usr/include/pangomm-1.4 -I/usr/lib64/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib64/glibmm-2.4/include -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/cairomm-1.0 -I/usr/lib64/cairomm-1.0/include -I/usr/include/sigc++-2.0 -I/usr/lib64/sigc++-2.0/include -I/usr/include/cairo -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libdrm -I/usr/include/libpng16 -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/util/str_util.cpp -o src/util/str_util.o
c++ -c -Isrc -I3rd_party -g3 -D_USE_MATH_DEFINES -DGLM_ENABLE_EXPERIMENTAL -fdata-sections -ffunction-sections -pthread -I/usr/include/yaml-cpp -I/usr/include/gtkmm-3.0 -I/usr/lib64/gtkmm-3.0/include -I/usr/include/atkmm-1.6 -I/usr/include/gtk-3.0/unix-print -I/usr/include/gdkmm-3.0 -I/usr/lib64/gdkmm-3.0/include -I/usr/include/giomm-2.4 -I/usr/lib64/giomm-2.4/include -I/usr/include/pangomm-1.4 -I/usr/lib64/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib64/glibmm-2.4/include -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/cairomm-1.0 -I/usr/lib64/cairomm-1.0/include -I/usr/include/sigc++-2.0 -I/usr/lib64/sigc++-2.0/include -I/usr/include/cairo -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libdrm -I/usr/include/libpng16 -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/pool/pool_manager.cpp -o src/pool/pool_manager.o

c++ -c -Isrc -I3rd_party -g3 -D_USE_MATH_DEFINES -DGLM_ENABLE_EXPERIMENTAL -fdata-sections -ffunction-sections -pthread -I/usr/include/yaml-cpp -I/usr/include/gtkmm-3.0 -I/usr/lib64/gtkmm-3.0/include -I/usr/include/atkmm-1.6 -I/usr/include/gtk-3.0/unix-print -I/usr/include/gdkmm-3.0 -I/usr/lib64/gdkmm-3.0/include -I/usr/include/giomm-2.4 -I/usr/lib64/giomm-2.4/include -I/usr/include/pangomm-1.4 -I/usr/lib64/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib64/glibmm-2.4/include -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/cairomm-1.0 -I/usr/lib64/cairomm-1.0/include -I/usr/include/sigc++-2.0 -I/usr/lib64/sigc++-2.0/include -I/usr/include/cairo -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libdrm -I/usr/include/libpng16 -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/pool/unit.cpp -o src/pool/unit.o
c++ -c -Isrc -I3rd_party -g3 -D_USE_MATH_DEFINES -DGLM_ENABLE_EXPERIMENTAL -fdata-sections -ffunction-sections -pthread -I/usr/include/yaml-cpp -I/usr/include/gtkmm-3.0 -I/usr/lib64/gtkmm-3.0/include -I/usr/include/atkmm-1.6 -I/usr/include/gtk-3.0/unix-print -I/usr/include/gdkmm-3.0 -I/usr/lib64/gdkmm-3.0/include -I/usr/include/giomm-2.4 -I/usr/lib64/giomm-2.4/include -I/usr/include/pangomm-1.4 -I/usr/lib64/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib64/glibmm-2.4/include -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2. [... cut ...] b64/sigc++-2.0/include -I/usr/include/cairo -I/usr/include/librsvg-2.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/libdrm -I/usr/include/libpng16 -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/resources.cpp -o src/resources.o
make: *** Keine Regel vorhanden, um das Ziel „.git/HEAD“, 
  benötigt von „src/gitversion.cpp“, zu erstellen.  Schluss.

```

artbody, 2018-12-07:
ok bin doch ein Schritt weiter gekommen, 
ich hatte das Projekt als zip downgeloadet.

Als git clone ... ging es ein Stück weiter
```
src/util/step_importer.cpp:2:10: schwerwiegender Fehler: Quantity_Color.hxx: No such file or directory
 #include <Quantity_Color.hxx>
          ^~~~~~~~~~~~~~~~~~~~
Kompilierung beendet.
make: *** [Makefile:565: src/util/step_importer.o] Fehler 1

```

artbody, 2018-12-07:
Noch ein Schritt weiter

im Makefile 
`CASROOT = /usr/lib64/opencascade-7.3.0`

aber jetzt wird es komisch
```
src/imp/imp.cpp: In Konstruktor »horizon::ImpBase::ImpBase(const horizon::PoolParams&)«:
src/imp/imp.cpp:44:47: Fehler: keine passende Funktion für Aufruf von »zmq::socket_t::connect(std::__cxx11::basic_string<char>&)«
         sock_broadcast_rx.connect(ep_broadcast);
                                               ^
In file included from src/imp/imp.hpp:18:0,
                 from src/imp/imp.cpp:1:
/usr/include/zmq.hpp:405:21: Anmerkung: candidate: void zmq::socket_t::connect(const char*)
         inline void connect (const char *addr_)
                     ^~~~~~~
/usr/include/zmq.hpp:405:21: Anmerkung:   keine bekannte Umwandlung für Argument 1 von »std::__cxx11::basic_string<char>« nach »const char*«
src/imp/imp.cpp:56:47: Fehler: expected primary-expression before »int«
         int fd = sock_broadcast_rx.getsockopt<int>(ZMQ_FD);
                                               ^~~
src/imp/imp.cpp: In Lambda-Funktion:
src/imp/imp.cpp:68:57: Fehler: expected primary-expression before »int«
                     while (sock_broadcast_rx.getsockopt<int>(ZMQ_EVENTS) & ZMQ_POLLIN) {
                                                         ^~~
src/imp/imp.cpp:68:57: Fehler: expected »)« before »int«
src/imp/imp.cpp:68:60: Fehler: expected unqualified-id before »>« token
                     while (sock_broadcast_rx.getsockopt<int>(ZMQ_EVENTS) & ZMQ_POLLIN) {
                                                            ^
src/imp/imp.cpp: In Konstruktor »horizon::ImpBase::ImpBase(const horizon::PoolParams&)«:
src/imp/imp.cpp:85:40: Fehler: keine passende Funktion für Aufruf von »zmq::socket_t::connect(std::__cxx11::basic_string<char>&)«
         sock_project.connect(ep_project);
                                        ^
In file included from src/imp/imp.hpp:18:0,
                 from src/imp/imp.cpp:1:
/usr/include/zmq.hpp:405:21: Anmerkung: candidate: void zmq::socket_t::connect(const char*)
         inline void connect (const char *addr_)
                     ^~~~~~~
/usr/include/zmq.hpp:405:21: Anmerkung:   keine bekannte Umwandlung für Argument 1 von »std::__cxx11::basic_string<char>« nach »const char*«
make: *** [Makefile:559: src/imp/imp.o] Fehler 1

```

carrotIndustries, 2018-12-09:
Dein zmq.hpp ist zu alt, siehe auch https://www.mikrocontroller.net/topic/417908#5187138
Inzwischen ist sogar in debian stable ne neuere Version drin...

artbody, 2018-12-09:
Danke erst mal 
Also neuere Versionen gibt es bei gentoo gerade nicht.
Werd mich dann wohl mit einem git clone *
abfinden müssen oder ich befasse mich mal mit einem local ebuild

artbody, 2018-12-10:
Ok Gentoo funktioniert
https://www.mikrocontroller.net/topic/417908#5653937

closed 2018-12-10 by artbody