horizon-eda/horizon #218: Building on Linux

Luz, 2018-12-12:
Building fails, because "python3" is not found. Could this be missing under the build-dependencies?

https://github.com/carrotIndustries/horizon/wiki/Building-horizon-on-Linux

carrotIndustries, 2018-12-12:
Yes, it is missing but no one noticed since it's included in most distribution because some other package already require python.

Which distro are you using?

Luz, 2018-12-13:
nixos :)

carrotIndustries, 2018-12-13:
Can you provide a command similar to the ones already on the wiki?

Luz, 2019-01-04:
Unfortunately not yet, since I am not able to build it yet. I have problems with opencascade.
```
g++ -c -Isrc -I3rd_party -I/opt/opencascade/inc/ -I/mingw64/include/oce/ -I/usr/include/oce -I/usr/include/opencascade -I/include/opencascade -I/usr/local/include/OpenCASCADE -g3 -D_USE_MATH_DEFINES -DGLM_ENABLE_EXPERIMENTAL -fdata-sections -ffunction-sections -I/nix/store/avz3dr6s3q0dsqbxxp4lvddv9idqx7q1-sqlite-3.24.0-dev/include -I/nix/store/az2x3prv1l5rk8ag9qcc3dn0lbl6pg1a-libyaml-cpp-0.6.2-dev/include -I/nix/store/3j9akav6nlwgq2dwmf3vbnpd5zsl06q8-util-linux-2.32.1-dev/include/uuid -I/nix/store/v7n95rvvbvmm1wnblb73nc8xz52rvk1p-gtkmm-3.22.2-dev/include/gtkmm-3.0 -I/nix/store/drysynjqn77irp8m4qinm3n4w2c2p862-gtkmm-3.22.2/lib/gtkmm-3.0/include -I/nix/store/v7n95rvvbvmm1wnblb73nc8xz52rvk1p-gtkmm-3.22.2-dev/include/gdkmm-3.0 -I/nix/store/drysynjqn77irp8m4qinm3n4w2c2p862-gtkmm-3.22.2/lib/gdkmm-3.0/include -I/nix/store/9sd73fvxb7rwsr5r1sppjmk6ycw6bm4z-glibmm-2.56.0-dev/include/giomm-2.4 -I/nix/store/1zl7q4g4bv3q4qi2ac52gg0vcyiqfmvh-glibmm-2.56.0/lib/giomm-2.4/include -I/nix/store/9sd73fvxb7rwsr5r1sppjmk6ycw6bm4z-glibmm-2.56.0-dev/include/glibmm-2.4 -I/nix/store/1zl7q4g4bv3q4qi2ac52gg0vcyiqfmvh-glibmm-2.56.0/lib/glibmm-2.4/include -I/nix/store/p09bakl6r1jmxdzp04wcqspwvwifkksi-glib-2.56.0-dev/include/glib-2.0 -I/nix/store/1ih5zwhrsh1wwlx449pbqc2vp4dgdfxk-glib-2.56.0/lib/glib-2.0/include -I/nix/store/dhs8a9cclbq8mg6q5pzw9kf3c53dhzlp-libsigc++-2.10.0/include/sigc++-2.0 -I/nix/store/dhs8a9cclbq8mg6q5pzw9kf3c53dhzlp-libsigc++-2.10.0/lib/sigc++-2.0/include -I/nix/store/c046j7byy5n9ig27wvdw8m3cw1k92myp-gtk+3-3.22.30-dev/include/gtk-3.0/unix-print -I/nix/store/c046j7byy5n9ig27wvdw8m3cw1k92myp-gtk+3-3.22.30-dev/include/gtk-3.0 -I/nix/store/kp9c2rk54ifz27zlchl0mxp6jwv6afni-cairo-1.15.14-dev/include/cairo -I/nix/store/a4yws990fgm97h24dxwg3im74aajyh88-freetype-2.9-dev/include/freetype2 -I/nix/store/a4yws990fgm97h24dxwg3im74aajyh88-freetype-2.9-dev/include -I/nix/store/jid2dh6gzmqkbxz66agd1dnkdmdjga3h-pango-1.42.4-dev/include/pango-1.0 -I/nix/store/akihfz98jg40ds1i70xad4lhgq0k4y1d-gdk-pixbuf-2.36.12-dev/include/gdk-pixbuf-2.0 -I/nix/store/3v662vlphnznx8ylyzvbkzhxsr1k0912-atk-2.28.1-dev/include/atk-1.0 -I/nix/store/rwxaxlfw7rds56h6x24wrx32xjhixm5x-atkmm-2.24.2-dev/include/atkmm-1.6 -I/nix/store/ffg0bflq757z80y1p2wp7qbxvybvifmr-cairomm-1.12.2-dev/include/cairomm-1.0 -I/nix/store/fy89jsxryaxxwsfl48a9x98hnw7mrhvf-cairomm-1.12.2/lib/cairomm-1.0/include -I/nix/store/qn9kjnhk9pn0a117ffzkyy1vagvrcpk2-pangomm-2.40.1-dev/include/pangomm-1.4 -I/nix/store/x59ywxqhnqy1nqi64kqycrb01pf4kf6p-pangomm-2.40.1/lib/pangomm-1.4/include -I/nix/store/grhhw1393d8cy3rj0qkl9v8ai16sbjmm-epoxy-1.5.2-dev/include -I/nix/store/ic6yg8bb56h1hf121dk0ww83gg05qmyk-librsvg-2.42.4-dev/include/librsvg-2.0 -I/nix/store/6nwwk1hz4krk25lb6xk1prhqx8zrcm63-zeromq-4.2.5/include -I/nix/store/iw9sg1lj0ahmgwzpwc73y6fbzs8kv79b-libgit2-0.26.6/include -I/nix/store/q9gmvd6qcjbdpyyd0dng9mhmz8sppq89-curl-7.61.1-dev/include -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/util/step_importer.cpp -o src/util/step_importer.o
src/util/step_importer.cpp:2:10: fatal error: Quantity_Color.hxx: No such file or directory
```
The problem is that the hxx file is not found, as it probably is not a standard file ending, which makes the usual automatic building for nixos more complicated.
Is it easy to make opencascade optional? Is it only used for 3D-preview?

atoav, 2019-02-18:
AFAIK opencascade is used for interpreting the step files for the 3D-preview, as well as the `Export step` functionality.

Luz, 2019-02-27:
The thing with opencascade was, that the the header files were in a subdirectory called "oce", which i solved by adding that directory by the following line:
`NIX_CFLAGS_COMPILE = "-I${opencascade}/include/oce";`
The problem did not appear at kicad, because they used cmake for the build proces.

So the next problem that i currently have is the following:
```
install flags: SHELL=/nix/store/i82x3x0yiijkgyqkzh8ni87gspas0f48-bash-4.4-p23/bin/bash gsettingsschemadir=/nix/store/6b4421cfpgk5clx5m5x4x3s24pifhygi-horizon-eda-20190227/share/gsettings-schemas/horizon-eda-20190227/glib-2.0/schemas/ install
make: *** No rule to make target 'install'.  Stop.
```
Any idea?

carrotIndustries, 2019-02-27:
> Any idea?

There currently isn't an `install` target, though that might change once we have a release.

For now, just copy `horizon-eda` and `horizon-imp`to the appropriate location.

Luz, 2019-02-27:
Thanks, building finally works! But something still fails:
```
(horizon-eda:6552): GLib-GIO-ERROR **: 23:23:45.075: No GSettings schemas are installed on the system
Trace/breakpoint trap
```
I also have to admit, that I am using the window manager called "awesome". Might be my current problem?

carrotIndustries, 2019-02-27:
> I also have to admit, that I am using the window manager called "awesome". Might be my current problem?

Very likely not.

Seems to be related to this one: https://github.com/NixOS/nixpkgs/issues/45445

Luz, 2019-02-28:
Thanks, its starting now. How do I get the icons (the three on top right) to work?
Here is what I have so far:
https://github.com/NixOS/nixpkgs/pull/56497

closed 2019-03-03 by Luz