horizon-eda/horizon #220: Circular courtyard support

fruchti, 2019-01-02:
[_PCB Design Optimization Starts in the CAD Library_ calls for a circular courtyard for mounting holes on p. 100](http://www.pcblibraries.com/downloads/Guidelines%21PCB_Design_Optimization_Starts_in_the_CAD_Library.asp). I couldn't find any other examples, but at least a circular courtyard is mentioned [here](https://www.pcblibraries.com/forum/library-expert-2016-03-released_topic1875.html). For circular parts like THT electrolytics, test pads, etc. it would be nice to have that possibility IMO.

carrotIndustries, 2019-01-02:
Courtyards can be arbitrary polygons, so I don't see any reason why this shouldn't work.

fruchti, 2019-01-02:
Sorry, I should have been more precise. I was referring to the courtyard expansion via parameter program which is limited to regions composed of lines right now. Minimally, this would require `set-polygon` to allow for other shapes than `rectangle`.

If arbitrary polygons with arcs were to be supported (e.g. to fully image cylindrical metal can packages like a TO-100), the current system wouldn't be suitable anymore as far as I can tell.

carrotIndustries, 2019-01-06:
0117851f8f60f3f20cf1a32ca1495f3a41947c91 adds a circle shape to set-polygon (pops diameter from stack)

fruchti, 2019-01-08:
Thanks! Looks fine to me and should be absolutely enough for the majority of the cases.

closed 2019-02-06 by fruchti