horizon-eda/horizon #221: Idea: Schematic editor alternative symbol selection and pin name display

fruchti, 2019-01-13:
There doesn't seem to be a way to choose a different symbol for a part after it has been placed in the schematic. A field similar to the one used for the packages in the board editor would be nice here.

This would be pretty useful in combination with the “Pin display” setting (which I really like, BTW), because right now settings other than “selected only” often cause the pin names to overlap. I expect this to be pretty common with symbols where there are multiple pin names.
[image]
Always using the maximum size for the symbol so all settings would fit seems excessive to me. Also, it would make the pin display setting less useful.

The next-obvious idea would be to add a rule that there must be a fitting symbol for each case. I'd propose to specify the suitable “pin display” settings for each symbol in the symbol editor. The default setting would be “all three”.

If the pin display setting is changed in the schematic editor and the current symbol was marked as unsuitable for the new setting, another symbol for the part could be chosen automatically. If there are multiple possibilities, Horizon could either prompt or use the symbol with the fewest possible pin display settings (to ensure it won't just always use the largest symbol).

I believe there's a nice possibility for some UX enhancement here, but maybe I'm thinking too complicated. The symbol could also be just changed manually after all.

carrotIndustries, 2019-01-19:
a8f5c8c3651b47266ca6e308053cdbd646aa87f6 adds a tool to change symbols

fruchti, 2019-01-20:
Thanks! Yeah, it probably is, since you won't do that very often.

closed 2019-02-06 by fruchti