horizon-eda/horizon #222: Change parts to alternative Packages on the fly

atoav, 2019-01-14:
I am not sure if I am actually overseeing something, but I am in the situation where I'd like to swap some parts to an alternative package, and right now the only solution I see is creating extra parts with the alternative package as a default.  

Because this kinda defeats the whole purpose of the alternative package, I assume it is just not implemented yet?

In an ideal world I would be able to (on a per-part-basis) switch to all alternate packages directly from the schematic or board editor. Is there a way to do this?

carrotIndustries, 2019-01-14:
> In an ideal world I would be able to (on a per-part-basis) switch to all alternate packages directly from the schematic or board editor. Is there a way to do this?

Perhaps that's not that obvious, but you can change a part's package in the board editor by selecting an alternate package from the package combo box in the property editor on the right side of the window.

atoav, 2019-01-14:
Ahhh cool. I was pressing <kbd>Spacebar</kbd> and searching for `Package` and found nothing and thought it didn't exist. Thanks : )

closed 2019-01-14 by atoav