horizon-eda/horizon #223: 3D Preview crashes Package Editor with certain step files

atoav, 2019-01-19:
## Description
In the package editor's 3D Preview I will get a crash (of both the 3D Preview window AND the Package Editor, With *Exit Code 3*) when importing certain files. I was unable to track down precisely which files it hates and which it likes. The file that crashes it, is a `.step` with just a cube in it, [get the step file here](https://gist.github.com/atoav/838e4e061e0144e8c4644d3fcc6f8bbb)  
[image]   
  
The line referenced in the Error message is to be seen here:  
https://github.com/carrotIndustries/horizon/blob/a8f5c8c3651b47266ca6e308053cdbd646aa87f6/src/canvas3d/canvas3d.cpp#L844

## My current 3D Workflow
I have two workflows for the generation of 3D models:  
1. For all the more complex parts I use [Alias Autostudio](https://www.autodesk.com/products/alias-products/overview), export a step, add colors in FreeCAD and rexport from there to step. This works as long as you make sure to stitch all surfaces to make a solid without holes and non-manifolds otherwise your model won't show up in horizon
2. For repetitive parts (e.g. like caps) that come in a thousand variants and have very simple shapes, I use Solidpython which spits out OpenSCAD code, which I then import in FreeCAD to color it (OpenSCAD supports colors, but FreeCAD doesn't understand them). From FreeCAD I export to step. This is awesome for generating parametric stuff, that only changes in some dimensions.

The cube.step was generated with with method number 2. I have many working models generated with this method an example would be [the WIMA MKS2 series](https://github.com/atoav/horizon-pool/tree/wima_mks2) branch in my fork of the pool. These work fine but this simple cube crashes the editor for some reason.

## Solution
This might be again an issue with the STEP importer, but the Error seem to originate from [a part that handles displaying](https://github.com/carrotIndustries/horizon/blob/a8f5c8c3651b47266ca6e308053cdbd646aa87f6/src/canvas3d/canvas3d.cpp#L844), so it could be something different. In an ideal world a crash of the 3D-Preview window shouldn't drag the Package Editor down with it, which can cause users to lose unsaved progress.

carrotIndustries, 2019-01-19:
>  The file that crashes it, is a .step with just a cube in it, get the step file here

Is this the correct file? It contains something similar to an electrolytic capacitor, not a cube.

Error 1285 indicates `GL_OUT_OF_MEMORY`.

carrotIndustries, 2020-10-08:
Feel free to reopen if you run into this again.

closed 2020-10-08 by carrotIndustries