horizon-eda/horizon #224: Program hangs on Xubuntu 18.10 system

graunchit, 2019-01-19:
The software builds successfully on several different versions of Xubuntu but always when opening a schematic or board design the program hangs with horizon-imp running at 100% cpu and the only way out is to kill the process.

carrotIndustries, 2019-01-19:
Try attaching gdb or strace to see where the imp gets stuck at.

graunchit, 2019-01-19:
It is a problem somewhere with the display driver. After changing from the Nvidia proprietary driver to Xorg-Nouveau driver the programs do not hang but there are no window borders or titles displayed. Switching to IceWm and everything is displayed properly with no hangs.

carrotIndustries, 2019-01-19:
You may try the opengl demo from `gtk3-demo` to see whether this is related to `GtkGLArea` or horizon in particular.

graunchit, 2019-01-20:
strace logs may make sense to someone
Google shows a few similar posts of this happening with Xfce + Google Chrome browser. It may be relevant, I don't know.

sometime after execve horizon-imp

`4820  getpid()                          = 4820
4820  clock_gettime(CLOCK_MONOTONIC, {tv_sec=4832, tv_nsec=425543339}) = 0
4820  sched_yield()                     = 0
4820  clock_gettime(CLOCK_MONOTONIC, {tv_sec=4832, tv_nsec=425591027}) = 0
4820  clock_gettime(CLOCK_MONOTONIC, {tv_sec=4832, tv_nsec=425613697}) = 0
4820  clock_gettime(CLOCK_MONOTONIC, {tv_sec=4832, tv_nsec=425636193}) = 0
4820  clock_gettime(CLOCK_MONOTONIC, {tv_sec=4832, tv_nsec=425658813}) = 0
`

1,500,000 + more lines of this
then, kill the process

`4820  clock_gettime(CLOCK_MONOTONIC, {tv_sec=4869, tv_nsec=211729052}) = 0
4820  clock_gettime(CLOCK_MONOTONIC, {tv_sec=4869, tv_nsec=211749665}) = 0
4820  clock_gettime(CLOCK_MONOTONIC, {tv_sec=4869, tv_nsec=211776853}) = 0
4820  clock_gettime(CLOCK_MONOTONIC, {tv_sec=4869, tv_nsec=211809443}) = 0
4820  clock_gettime(CLOCK_MONOTONIC, {tv_sec=4869, tv_nsec=211837997}) = 0
4820  clock_gettime(CLOCK_MONOTONIC, {tv_sec=4869, tv_nsec=211858763}) = 0
4824  <... poll resumed> )              = ? ERESTART_RESTARTBLOCK (Interrupted by signal)
4820  --- SIGTERM {si_signo=SIGTERM, si_code=SI_USER, si_pid=4697, si_uid=1000} ---
4824  restart_syscall(<... resuming interrupted poll ...> <ptrace(SYSCALL):No such process>
4824  <... restart_syscall resumed>)    = ?
4823  <... poll resumed> <unfinished ...>) = ?
4822  <... epoll_wait resumed> <unfinished ...>) = ?
4821  <... epoll_wait resumed> <unfinished ...>) = ?
4824  +++ killed by SIGTERM +++
4823  +++ killed by SIGTERM +++
4822  +++ killed by SIGTERM +++
4821  +++ killed by SIGTERM +++
4778  <... epoll_wait resumed> [{EPOLLIN, {u32=1342252928, u64=140287859042176}}], 256, -1) = 1
4778  recvfrom(36, "", 8192, 0, NULL, NULL) = 0
4820  +++ killed by SIGTERM +++
4776  <... poll resumed> )              = ? ERESTART_RESTARTBLOCK (Interrupted by signal)`

and, final trace taken from a different run of horizon because I couldn't get strace to do the above log and this one at the same time

`strace: Process 5170 attached`
`^Cstrace: Process 5170 detached`
`% time     seconds  usecs/call     calls    errors syscall`
` 99.87    4.190724           2   1896687           clock_gettime`
`  0.13    0.005360           2      2283           sched_yield`
`100.00    4.196084               1898970           total`

graunchit, 2019-01-20:
Horizon ran perfectly while testing with several different window managers so I purged and reinstalled minimum XFCE and Horizon is now running on Xfwm with no problems.
Two of my machines showed the same problem over the past few months with XUbuntu 18.04  and 18.10.

closed 2019-01-20 by graunchit