horizon-eda/horizon #228: Error opening pool: stod

atoav, 2019-01-29:
I get the following error with the new pool (layer_help) on the latest build (https://github.com/carrotIndustries/horizon/commit/1b1fe9f5f7b5066d571ddbc178d432e81eefa49b). After the error the program returns to the home window.

[image]

carrotIndustries, 2019-01-29:
This is caused by the new parametric data feature not properly validating the given values
51515ed2b8696c1dd3a4822ba3993da826185db6 fixes this.

In your particular case, you'll have to delete the file `parametric.db` in the pool and update the pool to regenerate it. The pool update will then tell you which parts have invalid parametric data.

atoav, 2019-01-30:
Thanks for your time, it still doesn't work:

1. I delete `parametric.db`
2. I start the 51515ed version of horizon
3. Update Pool (no errors show up)
4. exit the pool editor afterwards
5. Enter Pool editor again – _**Error opening pool** stod_ shows up

carrotIndustries, 2019-01-30:
3c164b34160135b1671465206c051c349fc41500 does away with `std::stod`

atoav, 2019-01-30:
Ah okay cool I will try. I exported the `select * from resistors;` to csv and removed all lines fitting to `^.{8}-.{4}-.{4}-.{4}-.{12},\d*\.\d*,\d*\.\d*,\d*\.\d*` and there were none left, so I assume there are no german formated numbers in there.

I will try 3c164b3 as soon as it compiled

atoav, 2019-01-30:
Okay very cool, seems to work now. I just saw the new parts browser and color me stoked. I am blown away, that stuff is cool!

I think I can close this one then?

closed 2019-01-30 by atoav