horizon-eda/horizon #229: SIGSEGV on multi-Symbol placement

q3k, 2019-01-31:
When placing a 'Generic 12-pin connector' from the default pool, a dialog pops up asking me to select a symbol (either 1x12 or 2x6). I select a Symbol and place it. Then, however, the same pop up shows again. If I cancel, I get a SIGSEGV:

```
Thread 1 "horizon-imp" received signal SIGSEGV, Segmentation fault.
0x00005596ef8df8dc in horizon::Symbol::Symbol (this=0x7ffc3bab5af0, sym=...) at src/common/layer_provider.hpp:7
7	class LayerProvider {
LEGEND: STACK | HEAP | CODE | DATA | RWX | RODATA
─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────[ REGISTERS ]─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
 RAX  0x5596f0009f28 —▸ 0x5596ef8f4960 ◂— push   r15
 RBX  0x7ffc3bab5af0 —▸ 0x5596f0009ef0 —▸ 0x5596ef8d5680 ◂— push   r14
 RCX  0x0
 RDX  0x0
 RDI  0x7ffc3bab5b28 —▸ 0x5596f1f8b650 —▸ 0x5596f1dc0e30 —▸ 0x5596f1a05d00 ◂— 0x3
 RSI  0x5596f0009ef0 —▸ 0x5596ef8d5680 ◂— push   r14
 R8   0x5deece66d
 R9   0x5596f241cc10 —▸ 0x5596efffeb48 —▸ 0x5596efbd14b0 (horizon::ToolAddPart::~ToolAddPart()) ◂— lea    rax, [rip + 0x42d741]
 R10  0x7f6ed5b496e0 ◂— 0x1
 R11  0x246
 R12  0x7ffc3bab5ad0 —▸ 0x5596efffbd78 —▸ 0x5596ef957f10 ◂— sub    rsp, 0x18
 R13  0x7ffc3bab5fa0 ◂— 0x2a48ad4ada6787e9
 R14  0x0
 R15  0x7ffc3bab5b38 —▸ 0x5596f1f8b650 —▸ 0x5596f1dc0e30 —▸ 0x5596f1a05d00 ◂— 0x3
 RBP  0x0
 RSP  0x7ffc3bab5a10 —▸ 0x5596f0009f28 —▸ 0x5596ef8f4960 ◂— push   r15
 RIP  0x5596ef8df8dc ◂— movdqu xmm1, xmmword ptr [rbp + 0x10]
──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────[ DISASM ]───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
 ► 0x5596ef8df8dc    movdqu xmm1, xmmword ptr [rbp + 0x10]
   0x5596ef8df8e1    movups xmmword ptr [rdi - 0x28], xmm1
   0x5596ef8df8e5    movdqu xmm2, xmmword ptr [rbp + 0x20]
   0x5596ef8df8ea    movups xmmword ptr [rdi - 0x18], xmm2
   0x5596ef8df8ee    mov    rax, qword ptr [rbp + 0x30]
   0x5596ef8df8f2    mov    qword ptr [rdi - 8], rax
   0x5596ef8df8f6    mov    qword ptr [rbx + 0x38], r15
   0x5596ef8df8fa    mov    rsi, qword ptr [rbp + 0x38]
   0x5596ef8df8fe    mov    rdx, qword ptr [rbp + 0x40]
   0x5596ef8df902    add    rdx, rsi
   0x5596ef8df905    call   0x5596ef8d52e0
───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────[ SOURCE (CODE) ]───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
    2 #include "layer.hpp"
    3 #include <vector>
    4 #include <map>
    5 
    6 namespace horizon {
 ►  7 class LayerProvider {
    8 public:
    9     virtual const std::map<int, Layer> &get_layers() const;
   10 };
   11 } // namespace horizon
───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────[ STACK ]───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
00:0000│ rsp  0x7ffc3bab5a10 —▸ 0x5596f0009f28 —▸ 0x5596ef8f4960 ◂— push   r15
01:0008│      0x7ffc3bab5a18 ◂— 0x4
... ↓
03:0018│      0x7ffc3bab5a28 —▸ 0x7f6ede55aee7 (g_object_unref+103) ◂— lea    edx, [rbx - 1]
04:0020│      0x7ffc3bab5a30 ◂— 0x0
05:0028│      0x7ffc3bab5a38 —▸ 0x7f6ede55aee7 (g_object_unref+103) ◂— lea    edx, [rbx - 1]
06:0030│      0x7ffc3bab5a40 —▸ 0x7ffc3bab5aa0 —▸ 0x5596f1938e10 ◂— 0x0
07:0038│      0x7ffc3bab5a48 —▸ 0x5596f1a05d00 ◂— 0x3
─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────[ BACKTRACE ]─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
 ► f 0     5596ef8df8dc
   f 1     5596ef95c58e
   f 2     5596efc582ab
   f 3     5596efbd964a
   f 4     5596efbd385e
   f 5     5596efb97bba
   f 6     5596efb07e38 horizon::ImpBase::handle_click(_GdkEventButton*)+584
   f 7     7f6ee1bd00cd
   f 8     7f6ee07839db _gtk_marshal_BOOLEAN__BOXED+107
   f 9     7f6ede55611d g_closure_invoke+413
   f 10     7f6ede568c1e signal_emit_unlocked_R+2798
Program received signal SIGSEGV (fault address 0x10)
pwndbg> Quit
pwndbg> quit
Detaching from program: /home/q3k/Software/horizon/horizon-imp, process 7582
[Inferior 1 (process 7582) detached]
q3k@anathema ~ $ 
```

I'm running commit 391a6ada685c123daa98ca08db4cee7b840783d5 on Gentoo.

q3k, 2019-01-31:
Hm, I tried compiling with `-ggdb` but couldn't get any more debug info than this.

carrotIndustries, 2019-01-31:
0835f8fc8eb9ca45698ba6bc1a2dfcc1797f7439 should fix this