horizon-eda/horizon #231: Checkout branches of pool

frmdstryr, 2019-02-07:
Somewhat related to #165, what are your thoughts on adding a way `git checkout` a remote branch of a pool? 

It'd be nice to be able to have a list of remotes for a pool then be able to select a remote, see the branches on that remote, and checkout a branch for testing.  

It'd also great to be able to merge into new local branches (so as to never "corrupt" the main repo)
```
git checkout -b <some-generated-branch> 
git merge repo/remote
```

I know it can be done using git directly but I think this would make it easier to share/browse parts that have not yet been merged into the main pool.

carrotIndustries, 2019-02-07:
Some details on the current implementation of horizon's git(hub) integration:

The actual pool itself isn't a git repository, instead all of the interaction takes place in the `.remote` directory in the pool. I implemented it this way to avoid messy merge conflicts. When merging parts in the pool manager, these are copied to `.remote` and the actual git magic happens over there. Upgrading (I'm open for suggestions on renaming this button) happens by effectively doing a `git pull` in the remote pool, updating its database, diffing it to the local pool on a per-item basis and asking the user which changes should be applied to the local pool.

Any idea on how to implement the suggests feature from a users POV? What should happen in case of conflicts?

All of this isn't made easier by the fact that libgit2 isn't exactly easy to use, since it's not as high-level as the git cli client. https://github.com/carrotIndustries/horizon/blob/master/src/pool-prj-mgr/pool-mgr/pool_remote_box.cpp#L835 Everything starting at that line till the end of the function is needed to do `git add` and `git commit`.

frmdstryr, 2019-02-08:
Thanks for the explanation.  I guess all I really am looking for is a way to pull in parts, etc, that are not yet in the main pool.  

Would the same technique work for pulling from parts from other branches/repos as well? Or what's the proper way to use parts from an open PR?

carrotIndustries, 2019-02-10:
> Or what's the proper way to use parts from an open PR?

Right now, this isn't support by the pool manager itself, you'll need to use git manually or just copy the files to your pool.

Technically, implementing this is doable, we'll just need to work out how the implementation should look like:
 - Check out branch in remote pool (basically just fighting with libgit2)
 - Update remote pool
 - Do the usual pool upgrade/merge process

The last step is where things get difficult: Adding new items is easy, but what to do in case of modifications to existing items? At this point, we don't know if that item was modified by the user or by the PR. Overwriting the item with the version from the PR might not be what the user wants.

Only supporting adding items from PRs would be a solution that keeps complexity manageable, but still covers all(?) open PRs.

frmdstryr, 2019-02-14:
Manually making the pool a git repo then merging in worked fine (for the few PR's I merged in anyways).

> The last step is where things get difficult: Adding new items is easy, but what to do in case of modifications to existing items?

How does the pool cache determine "out of date"? Do units, parts, etc... include any sort of version number? 

Would it be difficult to do a 3 way diff (remote pool to local pool and local pool to main pool)?  If both show a change for an item then give an option to choose which (or neither) to use.

carrotIndustries, 2019-02-16:
> Manually making the pool a git repo then merging in worked fine (for the few PR's I merged in anyways).

While this might have worked fine for you, there are cases that result in hairy merge conflicts. That's why I deliberately chose not to make the actual pool a git repo.

> How does the pool cache determine "out of date"? Do units, parts, etc... include any sort of version number?

Out of date just means that the item in the cache is different from the item in the pool. This is determined by diffing the json files. Not perfect, but at lease more useful than a line-by-line diff.

> Would it be difficult to do a 3 way diff (remote pool to local pool and local pool to main pool)? If both show a change for an item then give an option to choose which (or neither) to use.

That's out of scope for the pool cache window for now, maybe later.