horizon-eda/horizon #232: Board editor crashes on Gerber export if file lock can't be aquired

atoav, 2019-02-11:
It seems the Board editor crashes when I export a Gerber file that is currently locked by another application (e.g. gerbv or such). This might cause some grief if the work has not been saved.

carrotIndustries, 2019-02-13:
I'm at a loss on how this could happen since the whole export process is wrapped in a try/catch block. I also wasn't able to reproduce this.

What's the output if you run horizon from the mingw shell?

atoav, 2019-02-14:
Weird. It happened twice when I was working on it 3 days ago, but I cannot reproduce it now either. I used [QuickGerberRender](https://github.com/ThisIsNotRocketScience/GerberTools/tree/master/QuickGerberRender) to generate images while it happened.  

I close this because I couldn't reproduce, but will reopen if I can put a finger onto it.

closed 2019-02-14 by atoav