horizon-eda/horizon #233: Schematic & board snapshots

frmdstryr, 2019-02-14:
Is it possible to "snapshot" different revisions of a schematic and board?  

I think it would be a really useful feature to have a way to snapshot the project state (pool cache, schematic and board layout), tag it with a version number, then have a way to either view or revert back to old revisions.

atoav, 2019-02-18:
I do this manually with git and it works incredibly well. If your projects include more than "just" the horizon project this is the better approach, because you can include stuff like simulations, documentation, images, panel designs, 3D-models, code...

Git is not easy to use, but it totally pays of learning it IMO. How would you use that feature?

frmdstryr, 2019-02-20:
I was thinking of a version list in the project manager that when one is selected it would then open the schematic / board at that version.

Yes git or just copying the project folder also works, just thought it may be easier to use. Feel free to close this.