horizon-eda/horizon #234: Netline styles

Peta-T, 2019-02-19:
is it possible to use different netline style in scheme editor?

carrotIndustries, 2019-02-20:
Right now, the net line style is determined by the net itself (diffpair/bus) and isn't directly settable by the user.

If adding such an option, it might look like this:
 - Color will be selectable from a few options that can be set as user prefrence
 - Thickness as well (default, thick, very thick)
 - Maybe dashed/dotted (needs shader support)
 - These properties will be settable on a per-net basis
 - Net classes can specify these properties (i.e. make all power nets red)

Peta-T, 2022-10-04:
Can you please send me some example?
my test is in top_block.json but without any output
```
...
    "net_classes": {
        "2dcecfed-c01c-46d8-98ff-8b4c77d56194": {
            "name": "myclass",
            "Color": "#0000ff"
        },
        "7abc9cbd-14fe-4261-9016-d0356b1f25d7": {
            "name": "default"
        }
...
```
Thanks

carrotIndustries, 2022-10-04:
Well, my previous was just a braindump of how things could look like. Nothing of that got implemented so far.

Peta-T, 2022-10-05:
And can you please tell me if i can expected in near feature?
For begining is enought maualy edit net_classes in file.
Thanks